EFeru/hoverboard-firmware-hack-FOC #253: try to act as a servo system

landrisan, 2022-01-05:
Hello,
At low speed, the motors like stepping. They are nor running smooth. Use SPD_MODE via UART
I intend to do a servo system, adding one encoder, but because the movement is not smooth al low speed, I can't do thys.
Also, is it possible somehow to control the motors at independent speed? Like when I do "Send(20, 40)", the first motor to run at speed 20 and the other at speed 40.
Thank you

Candas1, 2022-01-05:
Hi,

You might want to look at this:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/139

landrisan, 2022-01-05:
Thank you, just saved my day. You know, from thys howerboards can get verry nice servo systems for automation projects, at a verry low cost.

landrisan, 2022-01-05:
Ok, now I may ask to much, but I thynk the job is a half done... Is it possible to send by UART a number of steps that I want the motor to move? I suppose that in the modified firmware , if I change the 360degrees to 1 degree and just sent a +5V pulse on the input, the system may act as a stepper/servo, right?

Candas1, 2022-01-06:
You might want to discuss this with @vamfun himself. I think this is also what he wanted to achieve.