EFeru/hoverboard-firmware-hack-FOC #255: Is there any posibility to the hall sensor for the corresponding motor?

danylog, 2022-01-17:
Hi!
I am making a scooter (I only use one wheel) and yesterday I burned up a transistor from the right part of the pcb where my motor was connected and I needed to change the side and now use the transistors from the left side. BUT, I already cut the hall cable of the left side. 
So, is there any possibility to use the right hall cable for the left motor?
btw I am using hovercar firmware

Thanks

Angelina

EFeru, 2022-01-17:
Should be possible. You can feed here the left hall information:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4770b13b2cfc3a952f17f6217ec254639d10a0f5/Src/bldc.c#L220-L222

angelina4848, 2022-01-17:
> Should be possible. You can feed here the left hall information:
> 
> https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4770b13b2cfc3a952f17f6217ec254639d10a0f5/Src/bldc.c#L220-L222

`    rtU_Right.b_hallA       = hall_ul;
    rtU_Right.b_hallB       = hall_vl;
    rtU_Right.b_hallC       = hall_wl;`

like that?

EFeru, 2022-01-17:
Yes

closed 2022-01-21 by EFeru