EFeru/hoverboard-firmware-hack-FOC #256: To build an RC 4WD transpotter

waythe, 2022-01-21:
### Your System (Plan)
- VARIANT: VARIANT_USART
- CONTROL TYPE: FOC
- CONTROL MODE: Torque
- What is your application?
    - 4WD transported controlled by RC via Arduino

Hi, thanks for your great firmware !!
I am planning to make an RC transpotter which is able to carry up to 100kg boxes so I am planning a 4WD one.
And I planned to use an Arduino to receive the RC signal and to control the transpotter, because I need it to control leds as well.
I've nearly collected enough accessories and am about to start.
however, I've got some issues that not been figured out yet, I'd appreciate it if some guide can be given.
some of my questions may be a bit rudimentary as I'm not particularly good at circuits :).

Q1: Is FOC-Torque a proper mode for my build? I noticed that some issues pointed out the torque maybe not enough at low speed, I only need low speed so shall I try other modes or there's already a way out?

Q2: one battery (from the hoverboard) probably is not enough for the 4 motors, so I guess I'll need two of them. how shall I connect the batteries to the board? 1 battery to 1 board? if so, the two batteries are managed separately, but I don't want to charge connectors, two electricity meters etc..,  and if I connect the batteries in parallel seems there's a battery balance issue? and I am not sure how I can select one board as the mainboard to manage (measure & charge) the battery.

Q3: similar to https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/42 In my understanding, the switch of the hoverboard is a 'soft switch',  the 36v power is always connected and the MCU is always powered. when the power on button is pressed, the MCU will enable all other circuits, is it correct? then if I need the two boards and my Arduino to be powered on & off at the same time, I need to implement some protocol to send commands to each other? is there some material that I can use to catch up, like which pins to use. I am expecting: 
1. use the power button to power on & off the whole transpotter as it works to the hoverboard before
2. the arduino should play some  led effect when turned on (this is easy) and off (this is not)
3. the RC can power off the transpotter remotely, but cannot power on it.

Q4: Is there a good place to get a 5v supply for the Arduino? 
Q5: Is there a recommended way to power the 5v LED strips? (I've prepared some DCDC converter from 36V to 5V since LED strips need higher current, but I wonder where should I connect to get the 36V power, I want the DCDC to be powered only when the transpotter is powered on, to save energy)?

Candas1, 2022-01-21:
Hi,

Some quick answers, but there might be more to discuss:
Q1 : the original transpotter implementation is using voltage mode and works well, but you can try with other modes
Q2 : I think having 2 batteries in parallel powering both boards is better, you then need a single charger, and you can't have only one board powering off because of low battery. If should work fine if the batteries have similar chemistry and capacity. You still need 2 buttons to turn the boards on or off. Turning off could be handled by uart communication if needed.
Q3 : no, when you press the on/off button, it opens the mosfet and it powers the board and the MCU. It could be that it he issue you mentioned about the second board is waking up as the MCU gets power from the first board through UART. I am not sure how he connected the ESP and the boards. Actually this is how the split boards work (the model with one board per wheel), they have a single on/off button.
Q4 : the picture on the main page indicates where you can find 5v and 3.3v and how much current you can draw from there
Q5 : how many Leds are you talking about, you cannot use sideboards ? They have 12v leds, and there is a custom firmware for it

Have you read the wiki ? There is a lot of information about different variants, battery management and protocols.

A few days ago I thought about creating a transpotter with 6 wheels without using an arduino, but this would mean the firmware should be enhanced with:
- simple master/slave using the existing sideboard protocol (turning off is not in my scope)
- streamlining the current transpotter variant so it's using the dual inputs feature ( adc/gametrak or RC remote would be possible)

landrisan, 2022-01-21:
@waythe
Hello,
Yesterday, I have build a toy for my kids, so they can play inside the house on long winter days. I used  a howerboard with the modified firmware, a playwood panel and two rotating wheels in front. The result is a mobile platform that can support at least 150kg. So, the control mode is FOC SPEED, connected via UART with an ESP32 board with WIFI and Bluetooth. I used an PS3 remote control connected via bluetooth, I made an agorithm to establish the maximum speed given by the position of the joystiks and the acceleration/deceleration, direction by variating wheels speed and one button for send 0 speed(brake). The result is a verry precise vehicle, smooth movement, it has enough power for 1hour of play. Because is SPEED mode, when no steer is applyed, the vehicle runs absolutly streight, no metter the obstacles of each wheel. I powered the ESP32 board from the hall sensors. If you need more details, just let me know. Also, you may find some ideas in my project.

waythe, 2022-01-22:
@landrisan 
what a cool daddy! I am also hoping that my kid would love to drive the transpotter as a toy, he loves everything that has wheels, he's always rushing to help us with the IKEA FRAKTA trolley. I don't like that trolley that's why I am making this new one.

>the control mode is FOC SPEED, connected via UART with an ESP32 board 
>I used an PS3 remote control connected via bluetooth
>I powered the ESP32 board from the hall sensors.
your work is a great reference, my build is quite similar that I'm using a ESP8266 with a PS2 RC receiver connected to it.
I've tested the RC works well, and I can have many buttons left to control the light effect.
I am planning to program several driving modes to support both single-hand driving, track mode(left joystick for left wheels, right for right), etc.

>I made an agorithm to establish the maximum speed given by the position of the joystiks and the acceleration/deceleration,
I was thinking to map linearly the joystick position and the input param to the hoverboard (speed), what else algorithm is needed to make it move smoother? (regardless the load)

>a mobile platform that can support at least 150kg
>a playwood panel and two rotating wheels in front
that sounds pretty good. 
I planned to use such a build too, then I thought 4 same wheels may look better and work more steadily so I've already bought several used hoverboards, I want to make full use of it but the power wiring and on/off switch stucks me.

waythe, 2022-01-22:
@Candas1 
Thanks a lot for your fast reply.

> Q1 : the original transpotter implementation is using voltage mode and works well, but you can try with other modes

I get it, I will try different modes to see how they work differently. but seems FOC is always suggested other than SIN?

> Q2 : I think having 2 batteries in parallel powering both boards is better
> If should work fine if the batteries have similar chemistry and capacity. 
> You still need 2 buttons to turn the boards on or off. Turning off could be handled by uart communication if needed.

I got the batteries from used hoverboards and I guess their remaining capacity should be different. I know it should be problematic if I use them in series, but should be safe and no much side effect if in parallel?

> Q3 : no, when you press the on/off button, it opens the mosfet and it powers the board and the MCU.

then maybe it will work if I connect one button to two boards, but in case the on/off status of two boards becomes inconsistent, I may need to prepare some switch for the battery as well, to cut off the power physically.

> the second board is waking up as the MCU gets power from the first board >through UART.

I can somehow understand that the MCU can get power from UART then wake up itself, but how it powers the whole board including the power components? could it be that MCU will always keep the switch mosfet on once it starts work, so if I power the MCU, the board will automatically on? is it possible to realize a software switch by using this?

btw, does the [poweroff() ](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/36953c6fe9ba90593d21150113459b0a4bb4e312/Inc/util.h#L113)function really poweroff  the whole board?
if so, if I am able to modify the protocol that to send not only steer & speed but also a signal of poweroff, then I can poweroff the board with external UART devices?

>Q4 : the picture on the main page indicates where you can find 5v and 3.3v and how much current you can draw from there
>Q5 : how many Leds are you talking about, you cannot use sideboards ? They have 12v leds, and there is a custom firmware for it

Thanks, I saw the points where I can get 5v, just want to verify that its quality is good enough to power the microcontroller, what I worry about is the ripple and electrical surge, which may break an MCU in case if not isolated well with the fluctuating load like motors.
talking about the leds, I will use 2 strips with around 50 leds (WS2812B) each, they are powered by 5V, one led may draw approximately 50mA when set to full brightness (so 5A for 100leds) and about 1mA when 'off', but I'll not use all leds with full brightness so I guess ~2A could be enough.

the difficulty I have is that I want to power on everything with one button (instead of two or more), and power off everything with the same button or an RC command. And when it is powered off, I want to save energy as much as possible (so the DCDC, LED, Arduino, won't consume any power). I don't have a clear idea to realize it now.
If I power the Arduino with the HALL cable, then I think I can somehow achieve that by connecting the power button of two boards together, and realise a protocal which I can send 'poweroff' command via UART.
but when there are LEDs, I think I will need a DCDC, and I cannot directly connect the battery 36V to it because if so the DCDC will always consumes energy. I guess there must be a point on the board that I can draw big enough current with 36V only when the board is powered on, I can measure it myself when I have my hoverboards delivered.

> Have you read the wiki ? There is a lot of information about different variants, battery management and protocols.

Oh yeah, as well as some open or even closed issues. I am impressed that you maintained so well documented wiki as well as the code. thanks a lot for your hard work!

> A few days ago I thought about creating a transpotter with 6 wheels without using an arduino, but this would mean the firmware should be enhanced with:
> simple master/slave using the existing sideboard protocol (turning off is not in my scope) to be able to daisy chain several boards
> streamlining the current transpotter variant so it's using the dual inputs feature ( adc/gametrak or RC remote would be possible)

both enhancements should be very useful. 
I actually also wanted to realize a dual input transpotter with both gametrak and RC , but it's quite a pity that I cannot find even a single gametrak in my country, it seems it is only common in the UK. and since I have the Arduino, I can realize the dual input by my arduino.
will you carry out your idea? can you consider adding the poweron/off as well if you'll work on it?

Candas1, 2022-01-22:
Yes, FOC is the default motor control type, and the default control mode used which is used by the transpotter is voltage mode.
I have been told the BMSs should prevent too much current flowing between 2 batteries if the capacity is different, but it's worth searching for more details online.
Yes when the MCU is powered on, [firmware](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4770b13b2cfc3a952f17f6217ec254639d10a0f5/Src/main.c#L195) turns the mosfet on.
If you check the code, [poweroff](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4770b13b2cfc3a952f17f6217ec254639d10a0f5/Src/util.c#L1520) just turns the mosfet off.

From which country are you ? In France I could find few gametraks. They must be some on ebay.

waythe, 2022-01-22:
> Yes when the MCU is powered on, firmware turns the mosfet on.
> If you check the code, poweroff just turns the mosfet off.

I see, then probably I can get the way I described to power on/off them together.

waythe, 2022-01-22:
> From which country are you ? In France I could find few gametraks. They must be some on ebay.
from China, I do found some on eBay but almost all sellers are from UK. the overseas delivery cost is 3 times or even more than the price of a gametrak lol
lucky thing is that I can find some cheap hoverboards here.

waythe, 2022-03-19:
Things go slow because I don't have much spare time in recent months.
But finally, I am managed to build the frame of my transpotter last night and tested that I can run it with two wheels successfully.
thank you all! 
I also modified the firmware a little so that I can power off the board with the RC control.
However I met a new hardware trouble when working on the other board. Hope someone has some experience.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/274

closed 2022-03-19 by waythe