EFeru/hoverboard-firmware-hack-FOC #258: 2 beeps (low pitch): ADC timeout - advice

nosemangr, 2022-01-26:
### Your System
- VARIANT: ADC
- CONTROL TYPE: Default
- CONTROL MODE: Default
- What is your application? - Electric Go Kart (hopefully)

Hi,
first of all I want to thank you all for contributing to this exciting project.

I am getting the "2 Beeps (low pitch) - ADC Timeout".
What does that mean exactly?
I have 2 regular pots, connected correctly and when I flash the board with another firmware (hoverboard-firmware-hack-bewegungsappar.at) it works without problems, both for steering and forward reverse.

I have flashed it with yours, done all the calibrations, used a terminal to see the potis change the values, but I can't get the 2 beeps to stop, and the wheels to turn. Even while I'm reading live the ADC values in my terminal, while turning the knobs, the wheels are not spinning.
What does "ADC Timeout" mean and if you have any advice for troubleshooting it.

Cheers
Athanasios

Candas1, 2022-01-26:
Hi.

Looks like a bad calibration.
Adc timeout means the firmware sees adc inputs lower than the minimum identified during the calibration minus a threshold. This is called adc protection and the firmware goes into error more. There are some details in the wiki. 

You can check the terminal while you do the auto calibration to see what happens.

nosemangr, 2022-01-26:
> You can check the terminal while you do the auto calibration to see what happens.

Thank you very much. It was indeed an error from my side with Pot calibration. When I hold the button for 5 seconds I don't get the "beep" which led me to believe it's not working, so I was trying manual calibration, which I botched quite nicely by setting completely wrong numbers.
Figuring out that the Auto calib works without the beep, solved the issue and it works as expected.
Thank you!

closed 2022-01-26 by nosemangr