EFeru/hoverboard-firmware-hack-FOC #259: How to control motors in skateboard mode via arduino using PWM Signal?

dominik34, 2022-02-05:
### Your System
- VARIANT: Skateboard
- CONTROL TYPE: Torque
- CONTROL MODE: via Arduino
- What is your application?
    - Longboard

I'm new to this and having issues controlling the motors via Arduino in Variant Skateboard?

Using analogWrite(pwm, some value between 128 and 255) looks like the motors are doing what they want. 

Is some calibration needed?

Candas1, 2022-02-05:
Hi,

Why do you use an arduino?
Why an arduino Usart variant would work best, please check the wiki.

dominik34, 2022-02-05:
Just for testing some things ;) UART should be better but I am confuse that simple PWM is not working correctly...

Candas1, 2022-02-05:
PWM variant works well with a rc receiver, used it a few weeks ago. But never used it with an arduino, you are probably doing something wrong. Are you sure you are using proper pwm? The wiring is ok?
It shouldn't be a matter of calibrating, erase the chip to try without the calibration.

dominik34, 2022-02-05:
I think the PWM signal isn't correct. Heard something about frequency....but don't know anything about it :( Wiring should be okay. You can't do much wrong there.

What I found out:
If the Signal has the value 128 motors are not or just slowly moving - with no torque. (as expected zero position - half of 255)
But if the value gets over or below it, sometimes the motors start spinning verry fast and other times the board beeps 4 times....(which means General timeout (PPM, PWM, Nunchuk) - btw do you know what "general timeout" exactly means?
The reason why I tough there's something wrong with the pwm Signal.

Anyway thanks a lot for the help and the fast response.

Candas1, 2022-02-05:
The firmware will generate a timeout error if pulses are not between 900 microseconds and 2100 microseconds for some time

Candas1, 2022-02-06:
https://forum.arduino.cc/t/servo-with-analogwrite/438505

dominik34, 2022-02-06:
Okay so it has something to do with the frequency. I also tried it with port 5 or 6 which are capable of 900Hz but with same results.

It would be interesting to know how to control it via PWM but in my case that's ok. Will try usart then :)
Tank you!

Candas1, 2022-02-06:
I think you should try with that library for servos instead of analogwrite.

scholzol, 2022-04-11:
I can confirm the topic with the servo library. I tested it with an Arduino nano with the **servo.writeMicroseconds(x)** command and it works. x should be in a range from 1000 to 2000. where 1000 is full backwards and 2000 full forward. 1500 is stop. Similiar to usual RC commands. Be careful and connect the ground of the hoverboard input to the arduino. I got strange wheel jumps, when I forgot that.
`
#include <Servo.h>

Servo myservo;  // create servo object to control a servo
// twelve servo objects can be created on most boards

void setup() {
  myservo.attach(6);  // attaches the servo on pin 6 to the servo object
}

void loop() {
  myservo.writeMicroseconds(1600);              // tell servo to go to position in variable 'pos'
  delay(15);                       // waits 15 ms for the servo to reach the position
}
`

scholzol, 2022-04-11:
one addition: I used the variant: PWM. May be this has an influence.

closed 2022-05-11 by Candas1