EFeru/hoverboard-firmware-hack-FOC #262: Hovercar stopped working

jeromebg, 2022-02-13:
Hi, first thanx for all the explanation and code for hacking hoverboards, great fun !
I followed the tutorial to flash the motherboard with the hovercar variant.
All worked fine during 5 minutes and then it stopped working. 
I thought the battery was empty so I plugged it for a full charge.
But now the motherboard won't lit up anymore.
I plugged the ST Link on it to check if it was still alive, but if I connect the stlink with the 3.3V pin (without the battery plugged) then the usb key is no detected by windows. If I remove the 3.3v pin then the st link key works all right ?
Is my motherboard dead and if yes a clue why ?
Thank you !

Candas1, 2022-02-14:
Hi,

Have you check the first few cases in the troubleshooting section of the [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Troubleshooting) ?
Seems like a component is shorted.

jeromebg, 2022-02-14:
Hi, yes I saw that section : 

The board is not turning on, you hear the power stage buzz noise but no melody

Can happen if you haven't [flashed](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Compiling-and-flashing-the-firmware) the chip properly
The chip might be defective, can you connect to it with [st-link utility](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/How-to-Unlock-MCU-Flash) ?
The dc-dc converter providing 3.3v to the chip might be defective

but no idea how to check that, the motherboard have been flashed correctly as the hovercar was working.
I agree it looks like a shortcut but how can I check it and repair if it's possible ?

Thank you !

Candas1, 2022-02-14:
[image]

You can check if one of those components are shorted. There are videos on YouTube explaining how to check this with a multimeter

jeromebg, 2022-02-14:
[image]
Thank you.
I watched several videos to understand how to check but I don't understand anything, I'm a noob in electronics ;)
Can you tell me on what number I should place the pins of my multimeter and what values I should find ? And if one of these mofset is dead can it be changed easely ?

jeromebg, 2022-02-28:
Thnx for your message landrisan, I checked the moto mosfets but they are all good. The short is somewhere else but I can't find where. I removed the AMS1117 but I still have the short between the the ground and vin on the motherboard...

Candas1, 2022-02-28:
What about the mosfets on the top of the board?
It should be a MJD127 and a RU1HL13L on your board I believe.

jeromebg, 2022-03-02:
Thx for your help Candas1 !
I tested all the mofsets.
The only one shorted is the AMS1117 location... (well the location is shorted between the vin and the ground).
No idea where it could come from...

mogielson, 2025-11-26:
Hi, maybe this will help somebody. I had the same issue. I desoldered components in the following order: tip127, ams1117, capacitors c4, c10, c19, c20, c21, c22, c23, but even with everyting off the mainboard, the 3.3V and GND pins were still shorted. Then I used the heat gun with reducer and took off the MCU. Finally no short circuit! I replaced it with stm32f103rct6 from aliexpress, programmed it and it works again. I assume that I fried it because of connecting the 3.3V from the st-link instead of powering the mainboard from the 36V battery.