EFeru/hoverboard-firmware-hack-FOC #263: motors not listening to reciever

2by4mike, 2022-02-14:
I'm using PWM with FOC_CTRL and SPD_MODE, I'm controlling it using an ar8000, I am using roboduren's online compiler so I can post any of the files needed if that would help.

Once the main board beeps after being turned on the motors spin once I give them throttle but ignore the input I'm giving it after its starts spinning. 
It could be interference from the motors spinning, I tried to reduce interference used a cat-5 wire between the right pwm control input and the receiver, so I was wondering what else I could do to reduce the effects of interference if that is what the problem is. 

I was also wondering what I could do to boost torque as I don't need it to move very fast but its spinning some diy tank tracks so I need all the power I can get.

Candas1, 2022-02-14:
Hi,

There are few more suggestions in the [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-PWM).
Is the torque not enough ? Speed_mode should be good as it will apply enough torque to reach de required speed.

Candas1, 2022-05-11:
Hi,

Have you solved your issues ?

closed 2022-06-19 by Candas1