EFeru/hoverboard-firmware-hack-FOC #264: Drive control with 10 pole pairs

rostik9810, 2022-02-14:
### Your System
- VARIANT: VARIANT_USART
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: SPD_MODE
- What is your application?
    - RC Platform with Raspberry Pi

How to control a PMSM motor with 20 poles (10 pole pairs)? I changed n_polePairs from 15U to 10U in BLDC_controller_data.c, but it does not work. The motor reaches maximum rpm and becomes uncontrollable. SIN control in VLT_MODE is fine. Is there a solution?

Candas1, 2022-02-14:
Hi,

Pole pairs are only used for rpm calculation.
You must have an issue with hall sensor sequence.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#troubleshooting

rostik9810, 2022-02-14:
> Hi,
> 
> Pole pairs are only used for rpm calculation. You must have an issue with hall sensor sequence. https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#troubleshooting

Tried all 6 hall combinations, doesn't work. Any other solutions?

Candas1, 2022-02-14:
Do you really have to use speed mode?
You can use voltage mode as a workaround.

rostik9810, 2022-02-14:
Yes, because when controlling for speed, the platform goes straight ahead. It helped to change the sign in line 2590, file BLDC_controller.c `PI_clamp_fixdt_l((int16_T)-rtb_Gain3, rtP->cf_nKp, rtP->cf_nKi,
                           rtDW->UnitDelay4_DSTATE_eu,
                           rtb_TmpSignalConversionAtLow_Pa[0],
                           rtb_TmpSignalConversionAtLow_Pa[1], rtDW->Divide1,
                           &rtDW->Merge, &rtDW->PI_clamp_fixdt_l4);
`

closed 2022-05-07 by Candas1