EFeru/hoverboard-firmware-hack-FOC #265: No commands acepted from Hoverino

Borst2, 2022-02-21:
Hello every one,

i have a problem to control the mainboard with my arduino. 
So here the story:
All worked fine and I could control the Hoverboard with the example sketch. Then I modified the serial command structure to activate cruise control over serial. This is also working fine. 
After that I added some features to the Arduino code and the problem was there. 
The mainboard doesn’t follow the commands steer and speed. The Feedback I receive on the Arduino serial monitor is always 0 and the motor also doesn’t turn. 
Turning the motor by hand and activate cruise control, the motor holds the rpm. So motor is still alive. 

I modified line 458 in main.c to send the raw data and it repeats the correct value I am sending with my Arduino. The .cmd value is still 0 and don’t follow the speed and steer sending with the Arduino. 

Feedback.cmd2           = (int16_t)input2[inIdx].cmd    -> always 0
—> Feedback.cmd2           = (int16_t)input2[inIdx].raw. -> correct value

So I did a roll back to the original code, but also no success. Still always zero. 

Where could be the problem?
Communication seems to work, because I have the correct values in input2[inIdx].raw, but the for some reasons it is setting the value to 0 in input2[inIdx].cmd.

Thanks for helping 

Borst

- VARIANT: USART; TQ, FOC
- Arduino Nano @ UART3

Candas1, 2022-02-21:
Hi,

Erase the chip and flash it again

Borst2, 2022-02-21:
Thanks.

Do you mean writing the firmware with vscode or is there an other way / command to erase the chip? 

Only flashing the firmware again, didn’t help.

Candas1, 2022-02-21:
Flashing again doesn't delete all flash memory.
You probably did a bad calibration by mistake. In st-link utility you can erase the chip fully

Borst2, 2022-02-22:
Thanks a lot. 
Erasing the chip with ST Link Utility did the job. 

All works fine, also the additional command to activate cruise control. 
I don’t understand why I don’t need to add the new command in the hoverino checksum calculation… but this is another issue. 

Borst

closed 2022-02-22 by Borst2