EFeru/hoverboard-firmware-hack-FOC #266: Hall effect ebike throttle

tadkari, 2022-02-24:
Im struggleing to connect and map my e bike hall efect throttle. Im a newby in this programming and code metter, so i menaged to turn off steering, but don't know how to change the code for ebike throttle and calibrate it. I know that it cant just be connected insted of a throttle pot, and used with pot and ADC right? So i need help with finding a whey to operate ebike throttle, because im thinking of using this type of controller for my 4x2 wheel mini E QuaD truck and Foc firmware and torque mode is working for me just fine. By the whey, thank you all for development of this wonderfull firmware, it is a great discovery for robotics and EVE enthousiasts.

Candas1, 2022-02-25:
Hi, have you checked the wiki?

tadkari, 2022-02-25:
Yes but, didnt find it.

tadkari, 2022-02-25:
I was looking at calibration section, maby i just need to calibrate the throttle?

Candas1, 2022-02-25:
Yes calibration will definitely help, you can use the autocalibration.
You will only have a throttle ? No Brake ?

tadkari, 2022-02-25:
Yes, for now im using only throttle, i have mechanical brakes and a freewheel on my transmisson, so brake wouldn't work anyway.
So i just need to calibrate the throttle? I was hooking throttle to adc set parameters for the pot, like any other noob.

tadkari, 2022-02-25:
When I set the input_1 of adc to 0, and input_2 to 3- i think 3 is automatic recognition, and hook the throttle, nothing happends, but works again with the pot

Candas1, 2022-02-25:
But have you done [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration#auto-calibration) ?
Have you read [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-ADC) ?
There is a youtube video, the guy is doing the autocalibration at some point in the video

tadkari, 2022-02-25:
Done these two things but not working, it doesnt recognise throttle, I put back a pot and it goes fine

Candas1, 2022-02-25:
Are you sure your throttle is not defective ?

tadkari, 2022-02-25:
Found a solution, it works with a throttle, i was missing a line in a code wich triggers the auto calibration. Thanks for your time.

closed 2022-02-25 by tadkari

Candas1, 2022-02-25:
but autocalibration is enabled by default, so I guess you have disabled it.

Good that you got it to work

tadkari, 2022-02-27:
No i did not, i just somehow had a code that was missing that line, it must have been a bug in compiler...

tadkari, 2022-02-27:
Im now facing with another issue, no metter what amperage i enter in the code for single motor, im stuck with 20Amps max, so how could i possibly overvome this problem? I know that the fets can hendle around 50, I putted thick layer of solder so that it can handle 30Amps easily, so mu question is, should i change the shunts also or just mess with thw code. For now i am using this controller as e bike controller and it saves the battery, and also the motor, but im in a steep hils area so standard 350w ebike controller with 30Amps max is no problem, but this one lowers the amps on steep hils and would not cross 20A.

Candas1, 2022-02-28:
Yes it's explained why it's happening at the end of this section
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits

tadkari, 2022-02-28:
Ive fully erased chip, reprogramed it, and than nothing, i couldn't turn it on anymore.

Candas1, 2022-02-28:
You probably didn't flash the chip properly.
Are you holding the power button as mentioned [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Compiling-and-flashing-the-firmware) ?

tadkari, 2022-02-28:
Yes, I did all that, but beeps came after importing other code, but foc wouldnt work several times, and then, after erasing chip and uploading the software finaly got it to work, but amps remain the same, also on steep hills it decreases amps and would come to a stop at 10 amps, full throttle. I set the current of a single motor to 30, and commutation curent to 33, calibrated it and nothing, still up to 20 amps.

Candas1, 2022-02-28:
My point is after auto-calibration, max current setting is persisted in eeprom, so changing it in config.h and flashing the firmware won't change anything. So you are saying it's still the same after erasing and flashing ?

If you have an ftdi, you could also activate [debug protocol](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Debug-Serial#debug-protocol), it would let you check and set max current without flashing.

Is there anything else that could limit the current ? What battery are you using ?

tadkari, 2022-02-28:
40 Amps of constant discharge , and 16Ah 36volt 10s battery im using, i menaged to get the amps higher trough your advices. Calibration helped after second try. Thank you for your time. Im greatfull. I menaged to programe two boards and set the parameters successfully . But third board in TRQ mode works roughly, and im hearing the crunching noise from stator windings, nither of the variants vorks how it should, in TRQ mode there is no free wheeling motor is stuck, works with really low rpm. It is a same chip GD32F103 RCT6. Maby an idea what it could be, it flashes normaly, but when i try it, it is not notmal as two other boards.

Candas1, 2022-02-28:
Have you tried to swap the phases?

tadkari, 2022-02-28:
Yes, several combinations, i changed the tip 127 tranzistor because board was turning on on its own when you connect the battery. 
So maby comutation is the problem? Drivers... Or something else?

Candas1, 2022-02-28:
But it's happening on both wheels?

tadkari, 2022-03-01:
Yes, on both wheels, and  im testing now only one side, the left one.

Candas1, 2022-03-01:
How is the board looking like, can you share a picture ?
Is it working well with commutation and sinusoidal control ?

tadkari, 2022-03-01:
No, its not, first two boards are green, rectangle shaped, and third one is blue.

> 
[image]
[image]
 The green one works floles

tadkari, 2022-03-01:
Its the same chip, programs well.

Candas1, 2022-03-01:
The board looks ok.
There can be many reasons, you need to check:
- different phase or hall mapping on this board
- defective port on the chip for hall sensor
- defective hall on the wheel
- defective mosfets

tadkari, 2022-03-01:
Phases are not different mapping, fets are ok, i replaced all 12 fets, wheel works without problems, so afther this mentioned, im left with defective hall port on the chip, i will check that.

tadkari, 2022-03-01:
Tried everything, but still programes/flashes well, works bad and rough and TRQ mode or any other mode couldnt work properly. So I'm saying its a faulty chip.

tadkari, 2022-03-03:
How can i change the direction of rotating for one of two motors?

Candas1, 2022-03-04:
Check hovercar variant in config.h, there are parameters for inverting singke wheel rotation, you can copy it to the variant you use

tadkari, 2022-03-04:
Oh thank you. You have been very helpful.
By the way i dismissed the blue motherboard because usart left was damaged, and hall sensors reading pins on the chip were gone. So i figured that only full working board can be used with this firmware. I menaged to programe one for my friend, and it works well with the TRQ mode. We are making some kick bikes with two motors so it's wonderful to reuse this controller.
Thank you again, and sory for my bad english, im from Eastern Europe, Balkan, Serbia.