EFeru/hoverboard-firmware-hack-FOC #267: speed value goes to zero when the value is changed via UART

scholzol, 2022-03-01:
### Your System
- VARIANT: USART3
- CONTROL TYPE:
- CONTROL MODE:
- What is your application?
    - testbed for a robocar

I'm trying to control the hoverboard with an Arduino nano and I'm using your hoverserial.ino. When I use this code everything works fine. The motors turning smoothly back and forth.

but when I change the code lines in the loop

```
void loop(void)
{ 
  unsigned long timeNow = millis();

  // Check for new received data
  Receive();

  // Send commands
  if (iTimeSend > timeNow) return;
  iTimeSend = timeNow + TIME_SEND;
  Send(0, iTest);

  // Calculate test command signal
  iTest += iStep;

  // invert step if reaching limit
  if (iTest >= SPEED_MAX_TEST || iTest <= -SPEED_MAX_TEST)
    iStep = -iStep;

  // Blink the LED
  digitalWrite(LED_BUILTIN, (timeNow%2000)<1000);
}

```
_____________________________________
to 
_____________________________________
 ```
void loop(void)
{ 
  unsigned long timeNow = millis();

  // Check for new received data
  Receive();

  // Send commands
  if (iTimeSend > timeNow) return;
  iTimeSend = timeNow + TIME_SEND;

  if(Serial.available()>0)
    { // only send data back if data has been sent  
    inByte = Serial.readString(); // read the incoming data
    iTest = inByte.toInt();
    }

  Send(0, iTest);
  // Blink the LED
  digitalWrite(LED_BUILTIN, (timeNow%2000)<1000);
}

```
--------------------------------------------
and send then values from Serial monitor of the arduino IDE, I get always stops (speed=0) between the changes. Any idea, where this behaviour comes from ?

br
Olaf

Candas1, 2022-03-02:
Hi,

As you can see [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#error-rejection) the firmware expects data every 160ms, is your code preventing that ?

scholzol, 2022-03-02:
not yet considered. Thx. 
But as you can see there is no delay in the code. May be, that the "send" from the serial monitor takes longer. Is there a problem to increase the timeout value from 160 to 300 for example ?

Candas1, 2022-03-02:
No you can increase it. This is a safety feature so the board stops when no commands are send, because the wires are cut for example.
Is the board beeping ?

scholzol, 2022-03-02:
Ok. Then I'll play around a litlle bit. I'll let you know, when I have some results. Thx again for your support.

scholzol, 2022-03-03:
So, I changed the value for SERIAL_TIMEOUT to 2000 for the moment and it works. I think I have to find an optimum for my command cycles and the timeout. for safety 2000 is to long. But it works. we can close the issue.

closed 2022-03-03 by scholzol