EFeru/hoverboard-firmware-hack-FOC #270: Sensorless drive

tadkari, 2022-03-09:
- VARIANT: ADC
- CONTROL TYPE: FOC
- CONTROL MODE: TORQUE
- Could this firmware drive the motors without hall effect sensors, sensorless?

Candas1, 2022-03-09:
No

closed 2022-03-13 by tadkari

tadkari, 2022-03-13:
Im using ADC input on my board, where can i putt the button for feald weakening, how can i even trigger it after programming and calibrstion?

Candas1, 2022-03-14:
Hi,

Current firmware doesn't handle this, code would need to be enhanced.
Can you create another issue as enhancement for that ? This one is about sensorless and is already closed. It will be hard for someone else interested in this topic to find it.

How would it work ? It would be active when the button is pressed ? It would be activated when pressing the button and deactivated when pressing it again ?