EFeru/hoverboard-firmware-hack-FOC #271: Help with board variant 2017.06.15 swagtron

greenmonster11, 2022-03-13:
I’m somewhat new to coding and circuitry so I apologize for my ignorance ahead.  I’ve been reading and watching videos and can not quite figure out what I’ve got wrong.  I have a 2 copies of a board variant that I can not seem to work out.  

It has a stm32F103c chip, appears very similar to all of the ones that I have come across.  It has a standard right sensor pin out, but the left appears mores similar to the AT32 version.

I also have a pre soldered RX TX GRND wire that I believe was from a Bluetooth module that gave me battery life, kid mode, etc.  .

I will add pictures, but main major question is is my board compatible?  

I am trying to map the pins but my limited chip experience is making it difficult.  Especially mapping the tx/rx pins from cable.

does this extra port work as an alternate input for usart 1 or is it independent.  Help determine proper pin out of large connector.   

I’m hoping someone has experience with these boards so I know if I’m wasting my time or not.   I appreciate all of your hard work on this.  
Thank you for any help.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Candas1, 2022-03-21:
Hi,

That's the first time I see such board.
As mentioned in the wiki you need to check if the pinout is matching and adapt in define.h if needed.

greenmonster11, 2022-03-25:
I'm still very new to this, I thought I would be able to dig through and figure it out but I decided to order the correct board and after I gain a better understanding of how everything works I will revisit these different boards.

> Hi,
> 
> That's the first time I see such board. As mentioned in the wiki you need to check if the pinout is matching and adapt in define.h if needed.

If you would like to check the board out further I have an extra one I can give you.  I'll send the sideboards and other components as well.  I can mail it to you if you are in the US.

I did not see a pm on github but this is my discord name.  
greenmonster#3697

Candas1, 2022-03-25:
Maybe you can try your chance on the [Telegram group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw), some other users might have such boards