EFeru/hoverboard-firmware-hack-FOC #272: ADC with feald weakening

tadkari, 2022-03-14:
- VARIANT:ADC
- CONTROL TYPE:FOC
- CONTROL MODE:TRQ

Im preparing to programe and calibrate the feald weakening, but i dont know where the button for engagement should be putt?

Candas1, 2022-03-15:
Did you read my reply? You didn't answer my questions

Candas1, 2022-03-15:
Hi,

Current firmware doesn't handle turning field weakening on with a button directly connected to the mainboard , code would need to be enhanced.
Another option that exists is by using a sideboard with [hovercar variant](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#features) 

How would it work with a button ? It would be active when the button is pressed ? It would be activated when pressing the button and deactivated when pressing it again ?
How would it work with cruise control ?

closed 2022-07-04 by Candas1