EFeru/hoverboard-firmware-hack-FOC #274: How can I judge whether a GD32F103 board is broken and possible to bring it back?

waythe, 2022-03-19:
Hi all, I am back with a new question lol 
(my previous question was here: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/256)
last night, I am finally managed to build the frame of my transpotter and tested that I can run it with two wheels successfully.

then I disassembled another hoverboard(which works before I disassembled it into pieces), soldered for pins to the programming holes (not immediately, but about 1hour after it disconnected with the battery, it is a GD32F103), and started to try to unlock it via ST-LINK (connected to the battery, kept pressing the power button while connecting, and connected SWCLK, GND, SWDIO to ST-Link, Did not connect 3.3.V)
however, I got a 'Can not connect to target!' error, no success when I change the reset type. then I suddenly realized that I didn't hear any beeps from the buzzer. 
I then tested that there's no 3v3 output from the empty programming hole.
I think it means the board(and the MCU) is not powered on successfully right?

I checked several times that there's no short circuit between the Pins I soldered, there's no obvious burning point on the board either.
Can someone give some hint on how I can test & diagnose the board and bring it back (if possible)? I cannot find an alternative in a short time lol.

[image]
[image]

waythe, 2022-03-19:
checked: the button has no problem, the battery has no problem.
any suggestion about checking the power supply circuit?

and I have another hoverboard with split GD32F130C8T6 boards, which is not compatible with this firmware.
I think I'd better not try using it mixed in my project right? the features/behaviors of two firmware may differ and cause trouble.

waythe, 2022-03-19:
Removed the battery and tried to flash the MCU with 3.3 v from the ST link.
but once plugin the 3v3pin, the ST link led indicator becomes off. and there is no power on melody from the buzzer either.
does it mean my MCU is dead? if so, can I fix it by replacing the MCU with a new GD32F130C8T6?
any other effects I can try?
(heard people mention tip127 from time to time, this is a tip127 051 on my board, how can I test whether it works?

I plug in the battery and the charger, the charger works (the LED indicates that it is charging), but I cannot power on the board still.

[image]
[image]

waythe, 2022-03-20:
I am lucky to be able to find a compatible second-hand hoverboard today.
I unlocked the motherboard, and flush the firmware successfully. (with the battery connected when flushing)
everything seems fine but when I tried to connect the board with the motor, it doesn't work (not powered on correctly)
I then tested the voltage between the battery + and -, it is zero with the board on, and 36v with the board removed.
so I guess somewhere has a short circuit but I cannot find it.
I tried again, the 78M05 burned with fire then.

now I am replacing the 78M05, but what is the reason for it lol?
I tried to be careful with the boards, and I always press the power button again or short circuit the board's XT60 after unplugging it from the battery to discharge the capacity. is it correct?

latest: I now replaced a new 78M05, then I measured the XT60 red (36v) is connected directly to the pin1 of 78M05, while the XT60 black to the pin3.
I tested my other board, it gives a different result 0L. now I dare not to connect the board with the battery without sufficient tests (the MCU didn't die yet). what else should I check?
maybe I can @Candas1 ?

waythe, 2022-03-20:
update:
fixed the second board by replacing 78M05 and two TIP127 (at least it can boot now)
the TIP127s were short. but I am curious why they die suddenly.

the 1st board I still have no progress yet.
I tested the ASM1117, the output is short with GND. I though it might be the ASM1117's problem.
but when I remove the ASM1117, the 3v3 is still short with GND.
Not sure yet where can be the problem, removing the MCU and test again may be the last choice (it is risky for me to solder so many pins).

Candas1, 2022-03-20:
Have you checked the mosfets?

waythe, 2022-03-20:
> Have you checked the mosfets?
just tested them. 
DIODE: S->D  0.535V, D->S half 0L, half 2.75V
BEEPS: G - S no beeps. 
3v3 - GND beeping

waythe, 2022-03-20:
I power the 3v3 with my adjustable power supply, with current limited to 1.5A.
the power supply enters CC mode, strange thing is that I waited for a while and no heating of any components was found.
The MCU is not getting hot either.

Candas1, 2022-03-21:
I broke a few boards this weekend during a combat robot competition, need to check if I have a similar case and what could be the reason.

waythe, 2022-03-22:
I removed the MCU of the 1st board, the short circuit still exists.
I tried to give a 5A current and still found nothing heated.
looks not like a simple capacitor failure. I guess my MCU might still be alive in this case but I have no way to identify it.
and I am very curious about what breaks the boards so that to avoid it in the future.

BTW @Candas1 
Is it practical to make 4wd cars by mixing the old one-mother-board(use EFeru's FOC firmware) and the newer split-mother-boards (let's say with GD32F130C8T6, which is suggested to use https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2)
I guess no because they may have different behavior even if I can unify the protocol?

how about mixing one-mother-board (use EFeru's FOC firmware) with  GD32F130C6 (which EFeru provides some other firewares)? do EFeru's firmwares share the same protocal/behavior?

Candas1, 2022-03-22:
I am very curious what's the issue. I fixed a board yesterday, I only had 2 mosfets of the same phase shorted so it was easy to fix.

Unfortunately the firmware for split boards is totally different. Not the same protocol and motor control.
But if you unify the protocol you could probably use all boards in commutation mode.

I started to investigate the split boards, there are many different ones, not all can do FOC from the hardware perspective.
I would like to try to port the firmware to those boards but I had no time yet. 

Someone tried to port the FOC model to this [fork](https://github.com/avenhaus/Hoverboard-Firmware-Hack-Gen2).
I created this [fork](https://github.com/Candas1/Hoverboard-Firmware-Hack-Gen2) that compiles with platformio.
FOC model wasn't working because it takes too much time to run probably. I need to try with a lower the PWM frequency.

Where are you located ?

waythe, 2022-03-23:
>I am very curious what's the issue. I fixed a board yesterday, I only had 2 mosfets of the same phase shorted so it was easy to fix.

That's lucky, I gave up fixing my first board, personally, I feel that its quality is better than the later one, its silkscreen matches the schematic diagram as well, but I have no luck with it.

>Unfortunately the firmware for split boards is totally different. Not the same protocol and motor control.
>there are many different ones, not all can do FOC from the hardware perspective.

I see, I know that matching different families of chips is tedious or even impossible.
I didn't expect much but still hoping to hear some exciting info though.
I am in China and the most convenient (also the cheapest) way for me to get wheels & monthboard is to buy second-hand hoverboards (whose battery may already be dead but the other parts functioning well). But you cannot distinguish the type of their motherboards before disassembling them. I've already had 2 split-boards hoverboards which I didn't expected. They are GD32F130C8T6 & GD32F130C6T6 as far as I can remember.
I am now finding some ideas to make good use of them. I can hardly create big builds since I have no room to store them.

> I started to investigate the split boards, there are many different ones, not all can do FOC from the hardware perspective.
> I would like to try to port the firmware to those boards but I had no time [...]

Good to hear that and thanks a lot for your contribution to the community :P 
platformio should be more friendly to beginners like me.
so what modes do those forks support? commutation & sinusoidal?

> it takes too much time to run probably
you mean the MCU is not powerful enough to compute in time?

closed 2022-03-26 by waythe