EFeru/hoverboard-firmware-hack-FOC #276: One pedal mode for person's with disabilities

Consti3D, 2022-03-22:
The idea is to have a torque mode that requires only one poti input to control the speed (one pedal mode in EVs) without the need of an input for the brake. Also a holding torque while standing still would be required to operate safely.
The purpose is to enable person's with disabilities to operate a vehicle with only one hand.

Candas1, 2022-03-22:
Hi,

Have you checked [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/2f6d1f191fd7470617eb14d4f6f17c729402e310/Inc/config.h#L517) ?

Candas1, 2022-03-23:
I added some of the information there to the [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR)
Please read it

Consti3D, 2022-03-23:
Thank you very much, I appreciate your responsiveness :) Did you have to modify the code or was everything in place already?I see you added Standstill hold and electric brake description. May I add that the behavior described there does not exactly match an EVs one pedal mode. In an EV the pedal does not have to be released to 0 to have an braking effect. Speed is already reduced when the pedal is released a bit so the braking force can be controlled. Maybe the parameter electric_brake_thres can be used to configure the behavior? Maybe you can add some more description about this parameter?Thanks!Consti

Candas1, 2022-03-23:
No everything was in place already, I just updated the wiki.
The description might be misleading, it was done a long time ago.
I think you need to try those features to understand how it works and if it can fullfil your needs.

Based on that we would understand if any improvement is needed.

Candas1, 2022-07-04:
Hi, any update on this topic ?

closed 2022-07-20 by Candas1