EFeru/hoverboard-firmware-hack-FOC #279: Read protection unlock problem

Consti3D, 2022-03-27:
Hi,
I'm struggling flashing the firmware. I'm using ST-Link v2 with firmware V2J39S7 and CubeProgrammer. I've connected GND, SWCLK, SWDIO and need to hold the power button to keep the board powered. After a lot of attempts I managed to clear the RDP option byte, however the flash memory has not been erased. From now on it is not possible to flash the program or full chip erase or reset the RDP. I'm kind of stuck, I already wasted a lot of time trying.
These are the errors I get:

> 23:31:14 : OPTION BYTE PROGRAMMING VERIFICATION:
  23:31:14 : Error: Expected value for STM32F1 Option Byte "RDP": 0xBB, found: 0x0
  23:31:14 : Error: Option Byte Programming failed

Did some blank checking and memory fill:
> 23:29:11 : Flash memory blank checking...
  23:29:11 : Time elapsed during the blank check operation is: 00:00:00.054
  23:29:11 : Warning: Flash memory is not empty at 0x08000000.
  23:29:27 : Error: The data size must be bigger than 0.
  23:29:27 : Error: Failed to fill memory with the given pattern
  23:29:36 : Error: Unable to read device id from ROM table

What is going wrong?

EFeru, 2022-03-28:
What kind of micro do you have, can you provide picture?

EFeru, 2022-03-28:
Have you tried the ST dongle with another micro? I think you need to work it out by elimination:
- confirm that dongle is ok
- confirm that wiring is ok
- confirm that micro is ok

Consti3D, 2022-03-28:
I can check if the dongle works on the sideboard STM.
For the wiring I checked GND and 3.3V. Also the chip info read out is successful so I would assume wiring is fine. Since the board powered down on connect I had to hold down the power button. How can I assure the board is not in calibration or broken calibration mode?

Candas1, 2022-03-29:
Autocalibration is a software feature of the firmware, to fail the autocalibration you need to be able to successfully flash the firmware.

I came accross some weird St-link dongles recently, but I didn't spend too much time on it as I had another working one, but I need to check if I am getting same issue.

What is also important for flashing is that you have good contacts/soldering.

Consti3D, 2022-03-29:
Yes I realized the USB side of the dongle is very loose and the ground shield is not soldered to the PCB.
Maybe also my wires are too long?
The software on the hoverboard for sure now is bricked because it's not possible to switch it on. When pressing the power button all lights are on for a second and that's it.
[image]
[image]
[image]

Candas1, 2022-03-29:
This means you were able to erase the chip and flash partially.
You just need to keep the button pressed when connecting or flashing the chip.

You need to sustain good connection during few seconds for flashing.

svenhaitjema, 2022-03-29:
if possible you could try reducing the programming speed. My boards will respond with the ID, but cannot be flashed above 0.25 Mhz, your cable's are quite long.

Consti3D, 2022-03-30:
Currently I'm in vacation but when I'm back I'll try long USB cable and short wires as well as slower communication speed. I think standard is 4mhz but I'll also try 0,25mhz.
Should I also solder the USB connector clamps to the PCB of the STM programmer or is there a reason why they are loose?
3.3V is not connected BTW

Candas1, 2022-05-11:
Hi,

Have you found a solution in the meantime ?

closed 2022-12-23 by Candas1