EFeru/hoverboard-firmware-hack-FOC #280: Field weakening setup for onewheel

svenhaitjema, 2022-03-29:
Hi All, 

First of all, the FOC firmware is awsome! I have build a OneWheel using 2 joined hoverboard motors. My first attempt with custom firmware was the bipropellant repo, after some fiddeling, I got it to work quite nice, but the motor is controlled with SIN(3rd harmonic) & phase advance. This gives a lot of speed, but the board is unable to balance itself past ~15 km/h. So I gave the FOC firmware a shot and copied over the onewheel logic from my bipropellant fork. The motors spin beautifully, but seem limited in speed. From the readme.md I understand that field weaking could help to speed things up without sacrificing torque too much, but some of the parameters are unclear to me.

This is the current motor config in config.h
```
#define CTRL_TYP_SEL    FOC_CTRL  
#define CTRL_MOD_REQ    TRQ_MODE
#define DIAG_ENA        1

// Limitation settings
#define I_MOT_MAX       15
#define I_DC_MAX        17
#define N_MOT_MAX       1000      

// Field Weakening / Phase Advance
#define FIELD_WEAK_ENA  0
#define FIELD_WEAK_MAX  5
#define PHASE_ADV_MAX   25
#define FIELD_WEAK_HI   1000
#define FIELD_WEAK_LO   750

```

1) In readme.md field weakening section, the picture shows the 'outside' setup, there field weakening is activated above a motor pwm value of 1000. Until now, it was my understanding that -1000 till 1000 is the valid range for motor pwm values. Does the range change when field weaking is enabled?

2) I am not really familiar with advanced drive modes like FOC + field weakening. Am I on the right track setting the control mode to torque + field weakening to control a onewheel or would you advice a different control mode, given the onewheel vehicle dynamics?

Candas1, 2022-03-30:
Hi,

Yes, [limits](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/bd4a34dabf655ba8a9a48e9974ebb0b619796fde/Src/util.c#L266) change when using field weakening.
But the way it's set by default, it's blended, so kicking in from 750 to 1000.

Field weakening will definitely help getting more speed.
Here a [discussion](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/86) with someone trying to reach maximum speed, there are few hints.

svenhaitjema, 2022-04-04:
Hi @Candas1,

Thank you for the reply! The use of field weakening does seem to increase the speed quite a bit :)

closed 2022-04-04 by svenhaitjema