EFeru/hoverboard-firmware-hack-FOC #281: Add a tip to the wiki page about correctly connecting motor hall sensors

ForcefulBlow, 2022-03-30:
Hello!

Recently I've been building a 2WD electric scooter, and I got quite frustrated when I tried to find a proper combination of phase/hall wires. 
It wasn't obvious if I connected hall/phase wires right in torque mode, because wheels were spinning with extreme speed no matter how lightly you touch the throttle (which is absolutely correct as it is a torque mode), and it was pretty jerky and sometimes it vibrated etc. 

However, in sine wave mode with voltage control it was extremely easy to understand if hall sensors were connected correctly.

Candas1, 2022-03-30:
Hi,

True, people could be using variant ADC first, as it's using FOC-voltage mode, to make sure the inputs and wheels are connected in the right way and working.
Hovercar variant/torque mode can be misleading without load, that's why I had added a video in the troubleshooting section of hovercar variant to show how braking behavior can look weird without load.

Regards,

Candas

Candas1, 2022-03-30:
Added an additional tip for [troubleshooting](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility/_edit#troubleshooting) in Firmware compatibility section of the wiki
@EFeru what do you think about that ?

closed 2022-04-17 by Candas1