EFeru/hoverboard-firmware-hack-FOC #282: Hardware Help - ADC showing different values on same boards

captFuture, 2022-03-30:
### YST Control board
- VARIANT: USART
- What is your application?
    - Custom Cart

I have a strange issue with 2 identical boards and the same flashed firmware.

- one board shows
in1:0 in2:0 cmdL:0 cmdR:0 BatADC:12 BatV:4655 TempADC:1683 Temp:304

- the other one shows
in1:0 in2:0 cmdL:0 cmdR:0 BatADC:1620 BatV:4275 TempADC:1667 Temp:302

With standard settings the second board was simply shutting down after a few seconds because of undervoltage 
I tried to identify hardware issues but did not find any broken components and unfortunately there is no schematic to be found for checking the right components. 

Did anybody have the same issue or has a hint for me how to go on?
Thanks

Candas1, 2022-03-30:
Hi,

Have you checked [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility)?
Any picture you can share ?

Candas1, 2022-04-17:
Hi, have you found a solution ?

closed 2022-05-07 by Candas1