EFeru/hoverboard-firmware-hack-FOC #283: Hoverboard not working after flash

AlaaImter, 2022-04-03:
I flash my hoverboard exactly like youtube videos and every thing ok ,,, but when i wire the potentiometer the wheel don’t move

AlaaImter, 2022-04-07:
Its not turn on

Candas1, 2022-04-08:
You won't get much help like this.
First you said wheels are not spinning, now that its not turning on
Have you read the wiki ? What have you tried ?

AlaaImter, 2022-04-08:
My friend 
I have 2 boards , first one flash and turn on and wheels not move with 1 potentiometer… when i try to use the second board when i flash it it’s turn off

Candas1, 2022-04-08:
And have you read [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Troubleshooting)?

AlaaImter, 2022-04-08:
First - i am use one potentiometer with 3v from board and the black and green wires from sensor is this true

Candas1, 2022-04-08:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-ADC#wiring

AlaaImter, 2022-04-08:
- “””If you don’t need steering, set input1_type to 0 in order to
   disable the input or go through the auto-calibration procedure”””

Can you please make it for me “ input1_type to 0”

Thirumaleswara, 2022-04-08:
my board has burnt connecting the potentiometer, now what to do ?

Thirumaleswara, 2022-04-08:
Hello Mr. Alaalmer,

I am curious to know the final answer. In the moment I can only guess what it could be. By the way can the question be limited to parameters values?

The problem is probably that those parameters are internally not systematically handled by a robust id, I suppose for reason of performance and robustness. Those input settings have a main and a delta value. In case of a running delta merge, they have two main and two deltas.

My idea would be to write a coding, that selects all the parameters that are duplicate, deletes them and then inserts them again without duplicates.

Has someone a better proposal?

KR
TR

AlaaImter, 2022-04-09:
I don’t know if i make i mistakes when the first time i make flash to hoverboard i run the board from the 3.3v st link usb , later i see its burn somethings

AlaaImter, 2022-04-09:
> 

Is there any way to make it on off button instead potentiometer?

closed 2022-04-10 by AlaaImter

AlaaImter, 2022-04-15:
> Have you done the auto calibration?
> It's explained in the wiki
Hello dear
Finally i run the wheels but on other firmware,, i like to try this firmware but after flashing with adc the board beeping I don’t why , i try many things but not working

Candas1, 2022-04-16:
Have you done the auto calibration?

AlaaImter, 2022-04-16:
I make it … i press power button about 8 seconds i hear a small beep but the beeping don’t stop and it’s not working…
Is there any thing in code must do it before calibration?