EFeru/hoverboard-firmware-hack-FOC #286: Beep when flashing with this firmware

AlaaImter, 2022-04-14:
After i uncomment adc and after flashing the bord only beeping and not moving… is there anything can i do

RoboDurden, 2022-04-17:
I guess it is better for you to build on VARIANT_HOVERCAR: The motors are controlled by two pedals brake and throttle. Reverse is engaged by double tapping on the brake pedal at standstill. See [HOVERCAR wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR).

Otherwise understand how speed is calclulated in main.c and find a free input to toggle it. If you don't need the steer adc, you could make a button pulldown/up that second adc input. Or go for the two 3.3V inputs of the short uart3 cable..

AlaaImter, 2022-04-17:
My friend i Uncomment hovercar and try adc variant , but until now i can’t run wheel … it’s beeping !!

RoboDurden, 2022-04-17:
Find the beep error codes in main.c (I guess).
Install the original Niklas Fauth Testmode Firmware (now lucysrausch)
Watch my beginners tutorials on YouTube.

AlaaImter, 2022-04-17:
lucysrausch firmware working good for me , but eferu not working

Candas1, 2022-04-18:
So is this an issue a question about how to reverse or is it about beeps you are getting ?

Have you checked what those [beeps](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics) mean ?
The [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-ADC) for ADC variant has a link to a nice video how to set up this variant end to end, have you watched it ?

This is not the first issue you are opening, you are making it very hard to help you. I have limited time/patience.

AlaaImter, 2022-04-18:
Sorry my friend 
But nothing working for me ,,, first issue the beeping after resolve it, reverse button

Candas1, 2022-04-18:
Well check what I have just shared.

RoboDurden, 2022-04-18:
@AlaaImter , when you have solved part of a problem you should tell how you did it so others get an idea of your skill level.
It is egoistic to only ask and ask for more and more .. If you want help from others you should help them in the best way to be able helping you..

AlaaImter, 2022-04-18:
I didn’t solve anything until now … i have 2 issues beeping and reverse button

Candas1, 2022-04-18:
After so many messages we still don't know what kind of beeps you are getting.
You might want to share a video of what is happening and how you run the calibration.

AlaaImter, 2022-04-18:
> After so many messages we still don't know what kind of beeps you are getting.
> You might want to share a video of what is happening and how you run the calibration.

In general this what i do
I wire only 1 wheel
I wire only 1 potentiometer 100k on green wire without solder any resistor on it
I uncomment adc variant
And comment sideboard_serial_usart3
And comment feedback_serial_usart3
And i flash it on board
And turn on … just beeping 
I try to make auto calibration but it’s not work

RoboDurden, 2022-04-18:
You need to disable the other  motor in config.h.
And you need to read..

Candas1, 2022-04-18:
But what kind of beeps ?????
Have you even clicked on this [link](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics) ?

Candas1, 2022-04-18:
Yeah @RoboDurden is right...

AlaaImter, 2022-04-18:
I am very sorry for you all … it’s working when i disable the other motor …

RoboDurden, 2022-04-18:
@Candas1 I have the same "problems" on my YouTube channel. When I get called "friend" I already know that they expect me to be their slave.

I like your wiki !
Maybe you should add a page with a help-protocol. With several steps like identifying the beep count and looking up the meaning  themselves .

Maybe an interactive help assistant might force them to check a few things before posting an issue.

But we should also understand ourselves that these firmwares target inexperienced users who barely speak English. The internet can make the whole planet feel as one :-)
Still i always try to turn people into better ones. And that means making them also care for other ones and stop being egoistic. Unfortunately these firmwares are mostly used for stupid fun projects (you might remember my usual rants.). Therefore stupid people insist on being egoistic.

Candas1, 2022-04-18:
Thank you.
I am spending a lot of time enhancing this wiki as I see some details are not covered.
I have structure it in the way that seems obvious to me, but it might not be straighforward for a beginner.
That's why I also create a [Troubleshooting](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Troubleshooting) entry point, but still the troubleshooting tips are spread everywhere.

I have also spend time recently making sure people know there is a wiki.
Now I am thinking people don't like to read, they want videos lol.

AlaaImter, 2022-04-18:
Mr candas 
After beeping resolved can i make a reverse by button ?

Candas1, 2022-04-18:
@RoboDurden already replied to that question

Imterhub, 2022-04-19:
Hello
After i solve beeping problem,,, i solder a resistor like this picture , but the wheel don’t turn on if i 
[image]
unsolder resistor it’s turn on …

RoboDurden, 2022-04-20:
Can not view the jpg.
I guess you need a 100k resistor from GND to ADC cable and a button from 3.3V to that ADC input.
And of course change the code in main.c to reverse speed when button is pressed..

Imterhub, 2022-04-20:
I test hovercar variant 
I use 1 potentiometer for speed and pedal for brake and reverse , the problem when i use brake the wheels spin backwards and forwards randomly and not stop ,,, only when the wheels stop permanently i can use brake and reverse …

Candas1, 2022-04-20:
[image]

Imterhub, 2022-04-20:
Thank you candas i will try it on ground, last question and maybe this is finally… can i increase the speed of hovercar i open main.c but I didn’t found from where i can increase the speed …

Candas1, 2022-04-20:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits#speed-limit

Protech2021, 2022-04-21:
I try to increase speed i change some value like this in picture but unfortunately the board shutdown, can i know where the problem 
[image]

Candas1, 2022-04-21:
Please check the rest of the wiki.

Protech2021, 2022-04-27:
> Can not view the jpg.
> I guess you need a 100k resistor from GND to ADC cable and a button from 3.3V to that ADC input.
> And of course change the code in main.c to reverse speed when button is pressed..

Can you explain more about this point button pressed in main.c ?

Candas1, 2022-04-27:
> > Can not view the jpg.
> > I guess you need a 100k resistor from GND to ADC cable and a button from 3.3V to that ADC input.
> > And of course change the code in main.c to reverse speed when button is pressed..
> 
> Can you explain more about this point button pressed in main.c ?

I helped you [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/289), do you expect a different answer ?

Protech2021, 2022-04-27:
> I helped you [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/289), do you expect a different answer ?

Yes mr candas
Here the question about normal button instead potentiometer and in main.c not util.c i think it’s difference

RoboDurden, 2022-04-27:
In all his posts he did not show a single little problem solving step himself ! Only like a little baby crying to get what it wants. 
I really like to help people. But that means i like to help people improving their own skills (so they can start helping others..)
But here i only see total ignorance in improving and only abuse continuing.

Protech2021, 2022-04-27:
> In all his posts he did not show a single little problem solving step himself ! Only like a little baby crying to get what it wants.
> I really like to help people. But that means i like to help [...]

You are a arrogant and unhelpful person… so please don’t reply to any questions for me

Thirumaleswara, 2022-05-06:
> 
> You are a arrogant and unhelpful person… so please don’t reply to any questions for me

What is the struggle for you in this? What is easy for you and what is more difficult?
What if you are already doing brilliantly at helping someone with right questions?

This is a hobby and not a professional support, people are trying to help you but have to be openminded polite and and run the last mile

Candas1, 2022-05-11:
I am closing this thread, this is going nowhere.

closed 2022-05-11 by Candas1