EFeru/hoverboard-firmware-hack-FOC #287: How to control one wheel per channel

TITAS1533, 2022-04-16:
### Before asking a question
See if it is already answered in [Wiki pages](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki)

### Describe your System so we can help you better
- VARIANT:PWM
- CONTROL TYPE:FOC
- CONTROL MODE:VLT
- What is your application?
    - RC platform

my plan is to make rover with hoverboard motors and control it with pixhawk but I need to control one wheel per channel?

Candas1, 2022-04-17:
Hi,

I implemented it my [fork](https://github.com/Candas1/hoverboard-firmware-hack-FOC) and I need to do a pull request but I was busy, maybe you can help test it ?
You need to use TANK_STEERING parameter.

TITAS1533, 2022-04-17:
I just commented out mixerfcn line in main.c and write
cmdR = speed
cmdL = steer

And it works fine

Candas1, 2022-04-17:
Ok that works as well.
If you still have a chance to test my fork, it will help having this feature without changing the code.
Thank you

closed 2022-04-17 by Candas1