EFeru/hoverboard-firmware-hack-FOC #288: TRANSPOTTER unkontrollierte Rückwartsfahrt

P-A-Trick, 2022-04-19:
TRANSPOTTER unkontrollierte Rückwartsfahrt!

Ist es möglich in der Programmierung die Rückwartsfahrt so einzustellen, so dass die Reifen nur blockieren/in der gleichen Position verweilen? Der Vorwärtsmodus klappt super, aber sobald die Leine von Gametrak nur wenige Zentimeter zurück ins Gehäuse fährt, macht der Transp Otter eine unkontrollierte Rückwärtsbewegung... 

Das ist leider mehr als Suboptimal...

Danke für den support!

Greetz from Cologne

P-A-Trick, 2022-04-20:
Leider ist das doch schon sehr gefährlich, wenn man teures Equipment damit transportiert. Fährt die Leine nur kurz ein paar Zentimeter zurück, schießt der TRANSPOTTER unkontrollierte weg. Man braucht meiner Meinung nach keine Rückwärtsfahrt, da der Wendekreis so klein ist und  man alles so gut steuern kann. Wäre es nicht möglich den Rückwärtsgang nur bei Bedarf durch Knopfdruck Kombi(z.B. 3x schnell an/aus Schalter drücken) Mir ist der TRANSPOTTER beim ersten Fahrversuch mit 2 Boxen für 2k € unkontrolliert abgehauen und hat die Leine abgerissen, dadurch ist ein GameTrak zerstört worden und der TRANSPOTTER kam erst nach 20m zum stehen..So etwas darf halt auf gar keinen Fall passieren.... Greetz

Mezgrman, 2022-04-20:
Das klingt aber so, als wäre was anderes noch zusätzlich kaputt. Ansonsten würde er, nachdem er ein paar cm rückwärts gefahren ist, selbst wieder stoppen, da die Leine ja wieder etwas weiter rausgezigen wird durch die Rückwärtsfahrt. Hast du Sicherheits-Pulldowns auf den Analogeingängen? Ich hatte etwas ähnliches mal, nachdem mir ein Poti gebrochen war.