EFeru/hoverboard-firmware-hack-FOC #289: backwards in Adc Variant

Protech2021, 2022-04-21:
How i can use backwards in Adc variant ??

Candas1, 2022-04-21:
Hi, have you checked [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration)

Candas1, 2022-04-21:
Ok you should have mentioning from the beginning.
This is not available as configuration in the firmware.
You would have to investigate and adapt the code.

Candas1, 2022-04-21:
but this switch has 3 outputs ? are those positions stable or going back to STOP if released ?

Protech2021, 2022-04-21:
It’s stable

Candas1, 2022-04-21:
Using [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/bd4a34dabf655ba8a9a48e9974ebb0b619796fde/Inc/config.h#L304) Parameter, you could have 2 switched connected to the right cable.

Then [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/bd4a34dabf655ba8a9a48e9974ebb0b619796fde/Src/util.c#L1041) , after button1 and button2 variables are populated with 0 or 1, you could do something like this:

if (!button1 && !button2){
  // forward switch and backward switch are OFF, it's the stop position, make the output 0
  input2[inIdx].cmd = 0;
}else{
 if (button2){
   // Backward switch is ON, multiply by -1 to reverse de direction, you could use -0.5 to also reduce the speed
   input2[inIdx].cmd *= -1;
 }else{
   // Forward is ON, Dummy code, there is nothing to do 
   input2[inIdx].cmd = input2[inIdx].cmd;
 }
}

Hope I understood your setup correctly.
It's untested so you need to be careful.

Protech2021, 2022-04-21:
I hope i can make it , the problem I don't have any background in programming

Candas1, 2022-04-23:
In adc variant, a voltage between 0 and 3.3v is measured, there results in a 0 to 4095 adc value (12 buts adc). Then this values is mapped to -1000->1000 if you are using a joystick like input.

Protech2021, 2022-04-23:
If i increase the 1000 value to 4000 is the speed increase too

Candas1, 2022-04-23:
Please read the wiki more carefully

Protech2021, 2022-04-27:
Here i get a error, support buttons right and serial usart3 not allowed, it’s on the same cable

Candas1, 2022-04-27:
Are you reading the comments ?
[image]

Imterhub, 2022-04-27:
thank you i disable it
i have a normal on/off button for test , it's 2 wires can i know where i connect it on red 3.3v and black ?

Protech2021, 2022-04-27:
Sorry i I didn't notice the note because i use the phone …
After i disable it ,,, i have a on/off button for test code the wiring on 3.3v and ground or something else ?

Candas1, 2022-04-27:
I don't remember 100% but I think those inputs are configured as pullup, so you need to connect it in such way:
GND->BUTTON1->blue wire
GND->BUTTON2->green wire

You should remove the existing code, it populates button1 and button2 variables.

Protech2021, 2022-04-27:
Thank you very much Mr candas everything is good …

Candas1, 2022-04-27:
I mean you shouldn't remove.
You tried?

Protech2021, 2022-04-27:
I removed (defined(SUPPORT_BUTTONS_LEFT)) and test it

Protech2021, 2022-04-27:
In adc variant can i use electric brake ?

Candas1, 2022-04-27:
If you need a brake why are you using adc variant?
Please read the wiki.

Protech2021, 2022-04-27:
hovercar have a brake but the reverse not easy for kid 3 years old to make double tapping for reverse, so i go to adc variant with on/off button for reverse

Candas1, 2022-04-27:
Why not sharing all those details from the beginning?

Protech2021, 2022-04-27:
From the beginning i am looking to solve reverse by button not pedal ,,, only this day i solve it with your help

Candas1, 2022-04-27:
For brake you need hovercar variant.
If you had mentioned from the beginning I could have helped to turn reverse ON with a button with hovercar variant, but now I feel I have just wasted my time.

Now you have reverse but no brake.

Protech2021, 2022-04-27:
This step (reverse by button) can go with adc variant and hovercar where the problem , now i have some experience to go to hovercar variant and try with it for brake

Protech2021, 2022-04-28:
Its work …

Protech2021, 2022-04-28:
Is there anyway to disable standstill before reverse work ?

if (button2){
here code to disable standstill 
input2[inIdx].cmd *= -1;

Protech2021, 2022-04-29:
?

closed 2022-05-15 by Protech2021

Germany111, 2022-05-15:
> For brake you need hovercar variant. If you had mentioned from the beginning I could have helped to turn reverse ON with a button with hovercar variant, but now I feel I have just wasted my time.
> 
> Now you have reverse but no brake.

I am looking for use push button switch to turn reverse