EFeru/hoverboard-firmware-hack-FOC #290: RPM Limit N_MOT_MAX is not working

pLazyN, 2022-04-21:
- VARIANT: ADC
- CONTROL TYPE: FOC
- CONTROL MODE: VLT/SPD/TRQ

Hi, thanks for all of your work and support for this project!
One issue I have is that the RPM Limit using the N_MOT_MAX variable in config.h seems to have no effect for me. No matter how low I set it, the speed in all modes stays the same as with the default value of 1000. Am I missing something? The rest of the settings is pretty much stock, other features i tried in config.h like FIELD_WEAK_ENA work like expected.

Candas1, 2022-04-22:
If you did the auto calibration, the value is saved in eeprom now. Please check in the wiki.

pLazyN, 2022-04-22:
Yes, I used auto calibration, but how is that related to N_MOT_MAX? Do I need to calibrate again after flashing in order to get the variable recognized? I thought just the ADC Limits are stored.

Candas1, 2022-04-22:
Current and speed limits are also stored:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits#gear-setting-up-the-current-and-speed-limits-on-the-go

Candas1, 2022-04-22:
I thought I missed to document it but it's also mentioned [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration#floppy_disk-saving-the-calibration)

pLazyN, 2022-04-22:
Oh, I have not checked to that feature as I did not indend to adjust the limits on the go. 

"will be stored in flash memory if you set them up on the go or if you do the auto-calibration"

What is the intention of the second case? So I cannot autocalibrate the ADC Limits without overriting the flashed N_MOT_MAX value? Or did I accidently enter the other mode by a second press and set the limit once to EEPROM? And now it is Set until erasing the flash or setting it again

Candas1, 2022-04-22:
Whenever we store data in EEPROM, we have to store all the data.
We cannot store only the limits for example.

At least this is how the firmware is developped.

closed 2022-04-22 by Candas1