EFeru/hoverboard-firmware-hack-FOC #291: UART getting command values but motor not moving

Kshmire, 2022-04-24:
First of all, thanks for your work and sharing it. I've been having a lot of fun and learning a lot from looking at your code.

### Describe your System so we can help you better
- VARIANT: USART
- CONTROL TYPE: 3 or 0 initially, later 2
- CONTROL MODE: UART
- What is your application?
    - Wheelchair

### Indicate what you have tried
Initially I had issues communicating with an Arduino Mega to the motherboard, always got 3 beeps.
I saw that the pinout to TX RX was different and I changed to pins 45, 46. But issue still existed.
Moved on to Arduino Uno using pins 2, 3 and then I was able to connect (I don't know why).

I have been able to get data from the SerialMonitor in the arduino IDE and I saw I could get the data properly, even if I manually moved the wheel, I could get the speed of it. But motors never move.

I also have been able to connect to the Web Serial Control with an FTDI and got the data properly again:
in1:0 in2:-15 cmdL:0 cmdR:0 BatADC:1508 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:-10 cmdL:0 cmdR:0 BatADC:1510 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:-10 cmdL:0 cmdR:0 BatADC:1514 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:-5 cmdL:0 cmdR:0 BatADC:1514 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:5 cmdL:0 cmdR:0 BatADC:1513 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:5 cmdL:0 cmdR:0 BatADC:1506 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:10 cmdL:0 cmdR:0 BatADC:1508 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:10 cmdL:0 cmdR:0 BatADC:1505 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:15 cmdL:0 cmdR:0 BatADC:1506 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:15 cmdL:0 cmdR:0 BatADC:1499 BatV:4020 TempADC:1629 Temp:408
in1:0 in2:20 cmdL:0 cmdR:0 BatADC:1513 BatV:4020 TempADC:1629 Temp:408

But still no motor movement at all, I can see in CMD2 that it's receiving from the arduino properly the speed (I change the increments in smaller ones, 5).

I have also tried autocalibration but without any input as right now I'm just using the arduino, later I will connect a joystick to the arduino to map the values to send.

What am I missing that makes the motor not spin?

Thank you for the support.

Candas1, 2022-04-24:
Hi, please check this. 
[image]

Kshmire, 2022-04-24:
Ok, I did not use a jumper so I was not linking my issue to the calibrated
values.

I also want to note down that I did an auto calibration without success.
But deleting the full chip with the STM flash utility and reflashing the
board did solve the issue!

Thank you very much for the tip. Now I can binge the fun part!

Candas1, 2022-04-24:
You're welcome

closed 2022-04-24 by Candas1