EFeru/hoverboard-firmware-hack-FOC #292: 4wd rover - PWM input

JAR4x4, 2022-04-25:
Hi, I'm building a rover with an additional aoutopilot. I need to control the four wheels individually as the autopilot will take care of the tank style skid steering.

Is individual PWM control supported by the firmware?

Alex

Candas1, 2022-04-25:
Hi,

It's available in the last update on my [fork](https://github.com/Candas1/hoverboard-firmware-hack-FOC), I am still testing but will commit soon.
Maybe you can help testing it.
You have to use TANK_STEERING parameter.
Why not using USART instead of PWM ? PWM is really sensitive to noise

JAR4x4, 2022-04-25:
@Candas1 , thanks for your reply!

I will test your fork in the weekend if i manage to assemble my test platform.

The ardupilot autopilot outputs PWM for each channel. I dont know how I can implement USART...

Candas1, 2022-04-25:
Is it motor pwm or servo pwm?

JAR4x4, 2022-04-25:
RC servo / ESC PWM

Candas1, 2022-04-25:
👍

Candas1, 2022-05-07:
Hi, it's now also available in this repository

closed 2022-05-07 by Candas1