EFeru/hoverboard-firmware-hack-FOC #294: Motor reverse in ADC mode?

KamilKaczy, 2022-04-26:
Hello! I have a question. I assembled electric scooter using hooverboard motors and mainboard, and i'm using EFeru firmware, but i have a problem, i was sure i assembled everything correctly, but it turns out that i made a mistake and front motor spins backwards :( I'm using ADC control mode, and in config file in ADC section i can't see any option to reverse one motor, and shafts of the motor itself are already welded into place.. Is there any possibility i could reverse it in firmware? I'm sorry if this is stupid question, i'm a bit of a noob here :)

Candas1, 2022-04-27:
Hi,

You should use [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Parameters#rewind-invert-wheel-direction)

KamilKaczy, 2022-04-27:
Thank You very much! I'll try it later today!

closed 2022-05-07 by Candas1