EFeru/hoverboard-firmware-hack-FOC #295: unpredictable wheel spinnings in PWM mode

scholzol, 2022-04-29:
### Describe your System so we can help you better
- VARIANT: PWM
- CONTROL TYPE: FOC
- CONTROL MODE: SPD
- What is your application?
    - 4WD with 2 hoverboards and RC control Graupner mx-20 + GR-24 receiver (ch1: speed; ch:2 steer)

all is running exactly as expected until I put the rover on the ground. This means: If the rover is free wheeling in the air, then all 4 wheels are turning in the right direction, when I move the joystick. And the wheels will stop, when the joystick is centered. But when I put the rover on the ground and try to drive, then unpredictable wheel spinnings occur and the rover doesn't stop immediately after releasing the joystick.

I made already the auto-calibration. I tried VLT mode. no difference.

Any idea, what can cause this behaviour ?

br
Olaf

Candas1, 2022-04-29:
Hi, have you checked in the wiki?

scholzol, 2022-04-29:
yes. I use the boards as described on the photo in the readme (mainboard_pinout.png). Is there any other possibility to test the correct phase and hall assignments with free spinning wheels ? Only with visual inspection everything looks fine. I'm not an expert in that field, thats why I'm asking.

Candas1, 2022-04-29:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-PWM

scholzol, 2022-04-30:
I assume, the wiring is ok. Because the rover is working correctly, when it's sitting on stand with free turning wheels. Do you know somebody, who has experience with PWM, I can ask ?

Candas1, 2022-04-30:
It's written there, pwm is noisy, you need to follow the recommendations there.

Candas1, 2022-04-30:
PWM is highly prone to noise : keep cables short, use 100nf capacitors, use shielded cable, use ferrites, stabilize voltage

Is it not clear?

scholzol, 2022-04-30:
yes, that I understand. But if the noise would be the problem, then I'd expect, that I get  that strange behaviour also when the rover is on a stand. This is the point I do not understand. Or am I wrong ?

Candas1, 2022-04-30:
The noise increases with speed/load

scholzol, 2022-04-30:
Ah, ok. now I understand. Then I'll proceed with the described measures. Can you give me an advise, where to place the condenser (near the board, I assume. Or on the receiver side). many thanks for your support.

Candas1, 2022-05-01:
Some scratch the pcb and connect the capacitors to ground after the resistors to create a rc filter. 
[image]

But improving the cable could be enough

scholzol, 2022-05-04:
I'll try this. First insert ferrits into the receiver cables. then possibly add capacitors on the board. Many Thx and sorry for the late answer.

scholzol, 2022-05-04:
Here some pictures from the PWM signals with the boards off and on

[image]
[image]

scholzol, 2022-05-05:
I used ferrits on the control cables, but no success. Can you tell me which size of capacitors you used (0402, 0603, ...) ?

Candas1, 2022-05-09:
Sorry I missed your reply.
This picture is not from me, it's from a member in the telegram group.
But I think 0603 would be ok.

scholzol, 2022-05-09:
Thx for your reply. Now I have to buy some pcs. Can take a while. I'll update as soon I have progress.

goatzillax, 2022-05-26:
Is there anything in the PWM code that kicks out inputs coming in faster than 50hz?

Candas1, 2022-05-26:
> Is there anything in the PWM code that kicks out inputs coming in faster than 50hz?

Read the wiki page about pwm variant

goatzillax, 2022-05-26:
I don't see in the Wiki where it states the code guards against pulses faster than 50hz, probably because it looks like the code doesn't guard against pulses faster than 50hz.

It has some generic guards for being IN_RANGE but that's about it.  There are also some strange things, like the PWM CH1 ISR checking the level of CH2 before it records the start time, which I'm not sure I quite understand.

Candas1, 2022-05-26:
Check again

goatzillax, 2022-05-27:
**No, the code does not check for duty cycle.**  If it were checking duty cycle it would at least check pwm_CNT_prev_ch1 before overwriting and forgetting it when the new rising edge came in.

I see now why the ISR is sampling the other channel's value now.  The ISRs are sharing (and resetting) TIM2 based on the assumption that receivers will essentially only have one active pulse across the channels.

This is...  fragile.

Candas1, 2022-05-27:
If you are experiencing issues, you can raise an issue, and we are also open to contributions if you find a fix.

But now I am not sure what is your problem, and what you are trying to solve.
This is an open source project. PWM code was contributed a long time ago, and seems to work for members in the community.
If I remember well, the current implementation of PWM is handling 2 channels with a single timer.

PWM variant is meant for RC receivers, this is described in the wiki:
[image]

goatzillax, 2022-05-27:
**The questions I am asking are not addressed in the Wiki.  Stop pointing to the Wiki.**

The wiki says what the code expects, it does not state what it checks.

The wiki does not explain that it actually expects certain behavior from PWM sources which is not necessarily guaranteed.

If you have two independent channels it's not actually spec'd that they essentially round-robin.  On some sources they act this way, but can't be guaranteed.  I don't think it's even officially spec'd behavior for RX's; they just happen to behave that way.

The problem is if you have noise on the line, the code handles it very, very poorly.

If a rising edge comes in, completes the pulse with another falling edge, then receives another rising edge before the 50hz period is up, it's possible that the last pulse and the next pulse are going to be garbage.  This would happen due to noise.

If I need to "look again", please specifically point out in the code where it checks that another rising pulse has not arrived within the specified period.  Go ahead and link it here.

Candas1, 2022-05-29:
> **The questions I am asking are not addressed in the Wiki.  Stop pointing to the Wiki.**
>
> The wiki says what the code expects, it does not state what it checks.
>
> The wiki does not explain that it [...]

I shared the wiki to be on the same page, a lot of people mistake rc/servo pwm and motor pwm.

As I mentioned, this is an open source project, this code was contributed long time ago. I contributed by adding IBUS variant as it's much more reliable, and that's what I am using.

If you think you can improve the pwm implementation, please contribute.

goatzillax, 2022-05-29:
Why are you asking me to "look again" for code that doesn't exist and pointing at a wiki that doesn't answer the questions?

Are you even using PWM at all?

Please point out the code that specifically checks for 50hz frequency on the PWM channels.  You clearly know how to paste links and screenshots, so go ahead and point out the code that you're saying does the check.

scholzol, 2022-05-29:
Hello goatzillax,

pls. don't use my thread for that topic. If you have an idea how the code can be improved, pls. open a new issue. I think it's an interesting idea you have, but pls. consider that this is an open source project and not a commercial software where you can send request to a supplier. It seems so, that you know what you are talking about. May be you find the part in the source code, where your idea can be integrated. Thx for your comments.

goatzillax, 2022-05-29:
Just because it's open source doesn't mean the other guy should lead everyone on a wild goose chase.

Did you solder the capacitors on the board?

Because I did, and it fixed the issue, but it's not something I would wish on my worst enemy.

What if I told you the problem you were seeing could have been fixed in software?

scholzol, 2022-05-29:
No I didn't solder the capacitors yet and I completely agree with you that this seems to be a terrible solution. As already mentioned in my former post, your proposed solution seems to be interesting. But let's look together (maybe with Candas1 and EFerus support) into the code and if possible improve it. But we shoukd create a separate issue for that.

goatzillax, 2022-05-29:
lol yeah it's like the worst solution ever.  I actually went ahead and did it just to validate the theory (although your waveform pretty much settled it).

If you're still thinking about putting the caps on here, since it's so tight in there, my suggestion would be to get some masking tape and a "fiberglass scratching pen".  Put a solder pad sized hole in the masking tape then put the masking tape on the PCB over where you think the pad should go.  Then use the scratching pen over it.  I haven't tried this yet, but I'd imagine this would work way better than trying to freehand with a hobby knife.  I went with 0805 sized caps since they're larger and a little easier to work with.  This will all depend on the layout of your board.

As far as software, I'm just going to drop this here and then walk away.

1. The PWM code should be looking at a global millis() or jiffies type counter using rollover safe math.  The timeout check should easily kick in before rollover issues.
2. Probably should stop cross-sampling channels because all that does is allow noise from one channel to essentially appear on the other channel.
3. The STM32 hardware apparently has a very basic input filtering stage in the hardware.  I can't tell if it's been turned on, but I'm pretty sure it would nail off a lot of the noise you're experiencing if tuned properly.

All this combined might slightly increase jitter, but I don't think +/- 1 or 2 ms of jitter really matters that much here, and it can't possibly be worse than the current state of the software and far preferable to exposing ground fill and physically soldering caps.

scholzol, 2022-06-15:
Now I soldered 100nF capacitors to each line as described above and it works now. It seems to be the most efficient way to eliminate noise from the lines. Many thx for the support.

closed 2022-06-15 by scholzol