EFeru/hoverboard-firmware-hack-FOC #296: RC PWM control

MarosMotaj, 2022-05-02:
- VARIANT: PWM
- CONTROL TYPE: RC_PWM
- CONTROL MODE: 
- Application:
    -  RC platform

Hi All thanks for great work! , I try to control 2W hoverboard with rc FrSKY Taranis transmitter but it does not work. The test Arduino program run perfect. Then I build and upload VARIANT_PWM to board. Board is quiet but motor are not react to transmitter. I use 3.3V to power receiver from flashing board pin and ground pin from right sideboard. I use CONTROL_PWM_RIGHT with this settings:

#define FLASH_WRITE_KEY       0x1105  
    #define CONTROL_ADC           0       
    #define CONTROL_PWM_RIGHT     1      
    #define PRI_INPUT1             3, 988, 1498, 2012, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define PRI_INPUT2             3, 988, 1498, 2012, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT1            3, 988, 1498, 2012, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT2            3, 988, 1498, 2012, 100  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section

First I try default firmware value scope, but with no reponse from wheels.... On transmitter I have above value scope. Thanks for help :)

Candas1, 2022-05-03:
Hi,

To make pwm work on the right cable, you just need to select variant_pwm, you don't have to change anything else. 
The values you have entered are wrong, the firmware is mapping from 1000<>2000 to - 1000<>1000 already. 
And you are sharing the settings for dual input? There primary input is adc, not pwm.

Please read the wiki.

MarosMotaj, 2022-05-03:
Hi,

Thanks for response :) Finally I find, that me receiver is faulty... I have used another one, which I connected to Arduino UNO. Than I modified the Arduino example code to be able to use rc receiver connected to Arduino UNO, and It works. :)

Thanks,

Candas1, 2022-05-03:
But why not connecting the receiver directly to the mainboard?

MarosMotaj, 2022-05-04:
Hi,

Because I used UART mode. Arduino is conncted via UART to mainboard. Receiver will be connected also via UART to Arduino so I avoid PWM signal problems generated long cable connections to mainboard.

Candas1, 2022-05-04:
Ok I understand.
I am closing the issue then.

closed 2022-05-04 by Candas1

MarosMotaj, 2022-05-04:
Ok, Thanks for help :)