EFeru/hoverboard-firmware-hack-FOC #297: Implementing external encoder

cohnhead66, 2022-05-02:
Hoping to implement an external rotary encoder to improve torque control a low speeds.  

### Describe your System so we can help you better
- VARIANT: USART
- CONTROL TYPE: Torque
- CONTROL MODE: USART
- What is your application?  Force Feedback, FFB),  Racing Simulator Wheel  

I have not been able to figure out how to utilize the wheel position provided by a rotary encoder to the BLDC controller to hopefully improve torque response.  

I am sending the rotation position of the wheel 0-360 as the steer command of the UART command structure.  I am sending the requested torque command as the speed command.  I am only using one wheel so the steer command is not needed in my application.  This information is sent from an external arduino that gets the wheel position from counting encoder ticks and the torque command by reading the PWM value provided by the FFB controller.

I need help in figuring out where in the code to copy the Steer value to rtU_Left.a_mechAngle that is needed in the bldc controller.

Also could use some help figuring out a calibration method to make the zero position of the external encoder match the zero position of the hall sensor.  I am planning to mechanically lock the wheel at the "hall sensor zero position", press a button to set external encoder ticks to zero, then unlock the wheel for use.

cohnhead66, 2022-05-09:
Quick update on where I am at with implementing an external encoder.

I have figured out how to update rtU_Left.a_mechAngle  with the angle from the encoder through the serial command.  I know this works because I am sending rtU_Left.a_mechAngle as one of the debug variables, and it tracks correctly.

Flipping the rtP_Left.b_angleMeasEna flag is doing something torque response becomes unusable.

I figure this is due to being out of sync, so I tried resetting zero point by moving the wheel in one degree increments, and resetting the encoder position to zero.  I tried this 30 times and never found that any degree of offset 0-30 degrees was in sync.  Since the hall sensors only provide 90 cts per revolution, I figured that they sync spot had to be within 30 degrees.

Can anyone think of a way I can get the estimated angle that the hall sensors are providing the controller, so i can compare to the encoder angle so I can make sure the two are in sync and see if the torque control works as expected.

Thanks

Candas1, 2022-05-11:
Hi,

Sorry I saw your message but I have actually no clue about this part...
@EFeru maybe you can help ?

landrisan, 2023-06-25:
Did you had any progress with implementing external encoder? Because I am verry interested too...

Candas1, 2023-06-25:
I don't think anyone got this working.

landrisan, 2023-06-25:
maybe GPT4 can help, if someone that knows the code will drive the AI. Because on the past month, I did some spectaculos codes using GPT4, thinks that would take days of work and research, I did in a few hours with AI help, I know the potential of the AI in the right hands

Candas1, 2023-06-25:
I don't think anyone is even trying.

landrisan, 2023-06-25:
Do you know who knows this code enough and could do it? Just let me know who can be and I will ask him. Would be a verry convenient way to do a servo sistem at low cost, high torque and low speed, completly customizable software inside the driver. A verry nice piece of work... I am ready to pay for this think....

korvolle, 2023-07-07:
Has this been solved yet? No one knows how to do it?

norlesh, 2024-05-07:
I know this is late to the thread put I'll put it out there anyhow -

The broad strokes on how to calibrate an encoder would be to start with b_angleMeasEna disabled during the first revolution of the motor, check within DMA1_Channel1_IRQHandler if the halls have changed and synthesize an absolute position value from the edges to store in a table along with the corresponding encoder value  as the endpoint parameters for an interpolation function that you would then use after enabling b_angleMeasEna to give accurate values to rtU_Left|Right.a_mechAngle
 ... an hour or two  with an LLM coder could fill in the blanks (I'm sick in bed so bags not it!)