EFeru/hoverboard-firmware-hack-FOC #298: Reading Bat.Voltage, Board.Temp data from board

MarosMotaj, 2022-05-04:
- VARIANT: USART
- CONTROL TYPE: RC
- CONTROL MODE: SERIAL_USART3
 RC platform

Hi,

Please, I am trying to get RX data from board (Bat Voltage, Board Temp) and it does not work. In Arduino I have uncomment #define DEBUG_RX. I define on Arduino "SoftwareSerial Hover1Serial(2,3);" RX on pin 2 and connected the pin to TX (blue) on right sideboard. When I run system I have "No data from RX". Where can be problem? I want to send Bat. Voltage and Board temp to display.... Thanks much :)

```
void Receive()
{
    // Check for new data availability in the Serial buffer
    if (Hover1Serial.available()) {
        incomingByte    = Hover1Serial.read();                                                       // Read the incoming byte
        bufStartFrame = ((uint16_t)(incomingByte) << 8) | incomingBytePrev;       // Construct the start frame
    }
    else {
        Serial.println("No data from RX");
        return;
    }

```

Candas1, 2022-05-05:
Please check the troubleshooting in the wiki before asking

MarosMotaj, 2022-05-06:
Hi,

I find in wiki how to do that and it works great on serial monitor web app :) It is possible to send the data to Arduino via direct "pin 2" connect and software serial method? "SoftwareSerial Hover1Serial(2,3);  // RX, TX" I try to do it but not working yet.... 

Thanks much :)

MarosMotaj, 2022-05-06:
Hi,

I am used ESP32 for telemetry data. I am connected ESP32 to left sideboard to get RX data and it works well. In future I will display the data on LCD. So I have Arduino UNO to controll the 4WD platform and ESP32 for LCD telemetry data view from two hoverboards.

Candas1, 2022-05-06:
Nice, don't hesitate to share a link to your project.

MarosMotaj, 2022-05-06:
Thanks much :)

I must buy another hoverboard because now I have only one, after I test and finish the project I will send you me project link :)

closed 2022-05-07 by Candas1