EFeru/hoverboard-firmware-hack-FOC #299: Auto power ON

landrisan, 2022-05-04:
Hello,
Is it possible that de mainboard to autostart on power connect, without the need of power button to be pressed?
I use in an industrial environement, on a complex machine and the need of manualy power on each board is a big problem for me

Candas1, 2022-05-06:
Hi,

Have you tried something already ?

landrisan, 2022-05-06:
Hello,
I did, but not a scientific method. I solved the problem for the moment by giving a pulse on power on pin from an external microcontroller. Still need  a better way...

landrisan, 2022-05-06:
If someone can help me with that... I will be gratefull

Candas1, 2022-05-07:
A hint from the troubleshooting section of the wiki for a hardware solution:
[image]

Another solution would be to replace the button with a jumper, but it requires changes in the code:
- commenting out [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4b16b647716ddc7ae03b0674b3b83bf304642c8f/Src/main.c#L209)
- commenting out the call to poweroffPressCheck function
- commenting out the calls to poweroff function might be needed also

landrisan, 2022-05-09:
Well, works perfectly. No more autopoweroff, works good. Used software soluution, not the TIP127. Was confused anyway, because one of my boards autostarts at power on, with original firmware, while other not. Now I know that was a hardware malfunction... Thank you.

Candas1, 2022-05-09:
Cool.
You know you don't have to fix this one lol

closed 2022-05-09 by Candas1

zeezooX, 2024-05-15:
> Hello, I did, but not a scientific method. I solved the problem for the moment by giving a pulse on power on pin from an external microcontroller. Still need a better way...

Can you tell me which pin did you send pulses to, the button pin or the latch pin?