EFeru/hoverboard-firmware-hack-FOC #303: Why throttle (hall effect) is reversed at ADC mode? Full speed at zero throttle

KamilKaczy, 2022-05-15:
### Variant

ADC

### Control type

Commutation

### Control mode

Voltage

### Description

Hi! For past 2 months i've build electric kick scooter using hooveroard hardware. And while mechanically i have everything sorted, last week i've been strugling with understanding how to make firmware work. My biggest problem right now, is that i have hall effect throttle (like in xiaomi m365 scooter) that is fed 3,3V, and outputs about 0,85V at zero throttle and about 2,5V at full throttle. My problem is that motors spin full speed at zero throttle, and close to zero speed at full throttle. As if throttle is reversed. Anyone knows what might be my problem? :( I just want to get it working. I tried inverting adc inputs and connecting throttle to adc2, same thing. Even with very short wires, and 3,3V regulator right next to throttle body. I have also added pulldown 1k resistor from signal to ground. It behaves weird, sometimes it doesn't even want to auto-calibrate, even tho throttle output is perfectly fine and linear and consistent, 0,85-2,5V. I don't know very much about firmware and coding, this is just a tool i really want to use to finish my project and utilize cheap hooverboard hardware, but it giving me such a headache, i tried changing some things in firmware, increased deadband, setting poti as "1" instead of "3" (auto detection), tried FOC as well as commutation method, torque and voltage mode etc.. Oh and i don't know if that means something to you, but when i touch signal wire with my hand, it changes the speed of motors to some degree. Any ideas what i could try? Should i share any more details or perhaps a video?
 If that might help you, here is my config:
 
 `// Define to prevent recursive inclusion
#ifndef CONFIG_H
#define CONFIG_H

#include "stm32f1xx_hal.h"

// ############################### VARIANT SELECTION ###############################
// PlatformIO: uncomment desired variant in platformio.ini
// Keil uVision: select desired variant from the Target drop down menu (to the right of the Load button)
// Ubuntu: define the desired build variant here if you want to use make in console
// or use VARIANT environment variable for example like "make -e VARIANT=VARIANT_NUNCHUK". Select only one at a time.
#if !defined(PLATFORMIO)
  #define VARIANT_ADC         // Variant for control via ADC input
  //#define VARIANT_USART       // Variant for Serial control via USART3 input
  //#define VARIANT_NUNCHUK     // Variant for Nunchuk controlled vehicle build
  //#define VARIANT_PPM         // Variant for RC-Remote with PPM-Sum Signal
  //#define VARIANT_PWM         // Variant for RC-Remote with PWM Signal
  //#define VARIANT_IBUS        // Variant for RC-Remotes with FLYSKY IBUS
  //#define VARIANT_HOVERCAR    // Variant for HOVERCAR build
  //#define VARIANT_HOVERBOARD  // Variant for HOVERBOARD build
  //#define VARIANT_TRANSPOTTER // Variant for TRANSPOTTER build https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter https://hackaday.io/project/161891-transpotter-ng
  //#define VARIANT_SKATEBOARD  // Variant for SKATEBOARD build
#endif
// ########################### END OF VARIANT SELECTION ############################

// ############################### DO-NOT-TOUCH SETTINGS ###############################
#define PWM_FREQ            16000     // PWM frequency in Hz / is also used for buzzer
#define DEAD_TIME              48     // PWM deadtime
#ifdef VARIANT_TRANSPOTTER
  #define DELAY_IN_MAIN_LOOP    2
#else
  #define DELAY_IN_MAIN_LOOP    5     // in ms. default 5. it is independent of all the timing critical stuff. do not touch if you do not know what you are doing.
#endif
#define TIMEOUT                20     // number of wrong / missing input commands before emergency off
#define A2BIT_CONV             50     // A to bit for current conversion on ADC. Example: 1 A = 50, 2 A = 100, etc
// #define PRINTF_FLOAT_SUPPORT          // [-] Uncomment this for printf to support float on Serial Debug. It will increase code size! Better to avoid it!

// ADC conversion time definitions
#define ADC_CONV_TIME_1C5       (14)  //Total ADC clock cycles / conversion = (  1.5+12.5)
#define ADC_CONV_TIME_7C5       (20)  //Total ADC clock cycles / conversion = (  7.5+12.5)
#define ADC_CONV_TIME_13C5      (26)  //Total ADC clock cycles / conversion = ( 13.5+12.5)
#define ADC_CONV_TIME_28C5      (41)  //Total ADC clock cycles / conversion = ( 28.5+12.5)
#define ADC_CONV_TIME_41C5      (54)  //Total ADC clock cycles / conversion = ( 41.5+12.5)
#define ADC_CONV_TIME_55C5      (68)  //Total ADC clock cycles / conversion = ( 55.5+12.5)
#define ADC_CONV_TIME_71C5      (84)  //Total ADC clock cycles / conversion = ( 71.5+12.5)
#define ADC_CONV_TIME_239C5     (252) //Total ADC clock cycles / conversion = (239.5+12.5)

// This settings influences the actual sample-time. Only use definitions above
// This parameter needs to be the same as the ADC conversion for Current Phase of the FIRST Motor in setup.c
#define ADC_CONV_CLOCK_CYCLES   (ADC_CONV_TIME_7C5)

// Set the configured ADC divider. This parameter needs to be the same ADC divider as PeriphClkInit.AdcClockSelection (see main.c)
#define ADC_CLOCK_DIV           (4)
[... 657 lines cut ...]
#if defined(SUPPORT_BUTTONS_RIGHT) && (defined(CONTROL_NUNCHUK) || defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT) || defined(DEBUG_I2C_LCD))
  #error SUPPORT_BUTTONS_RIGHT and (CONTROL_NUNCHUK or CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT or DEBUG_I2C_LCD) not allowed. It is on the same cable.
#endif

#if defined(CONTROL_NUNCHUK) && (defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT) || defined(DEBUG_I2C_LCD))
  #error CONTROL_NUNCHUK and (CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT or DEBUG_I2C_LCD) not allowed. It is on the same cable.
#endif

#if defined(DEBUG_I2C_LCD) && (defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT))
  #error DEBUG_I2C_LCD and (CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT) not allowed. It is on the same cable.
#endif

#if defined(CONTROL_PPM_RIGHT) && defined(CONTROL_PWM_RIGHT)
  #error CONTROL_PPM_RIGHT and CONTROL_PWM_RIGHT not allowed. It is on the same cable.
#endif

// Functional checks
#if (defined(CONTROL_PPM_LEFT) || defined(CONTROL_PPM_RIGHT)) && !defined(PPM_NUM_CHANNELS)
  #error Total number of PPM channels needs to be set
#endif
// ############################# END OF VALIDATE SETTINGS ############################

#endif

`

EFeru, 2022-05-16:
Hi,
You said you are using 1 kOhm pull down resistor. I think that is too low, maybe in the same range as your hall input (usually around 10 k ohm). Recommended value is 100 kOhm, see here:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR

That can explain the strange behavior and that wheels still spin when you touch the wires.

KamilKaczy, 2022-05-26:
It works like a charm. But only in hoovercar mode, not in ADC mode, weird because hoovercar used ADC as input, but well ok :)

KamilKaczy, 2022-05-28:
Hi Candas1
Yes i've been reading wiki multiple times by now :) 
And yes, indeed they shouldn't.
I compliled using Robo Durens online compiler as it was the simplest and and quickest way, and didn't required installing additional software

Candas1, 2022-05-28:
> Hi Candas1
> Yes i've been reading wiki multiple times by now :) 
> And yes, indeed they shouldn't.
> I compliled using Robo Durens online compiler as it was the simplest and and quickest way, and didn't required installing additional software

There is another way to compile without any software, it's described in the wiki, I even made a video

Candas1, 2022-07-04:
Hi, could you solve your problem ?

closed 2022-12-23 by Candas1