EFeru/hoverboard-firmware-hack-FOC #304: How to deal with the delay between UART commands?

aLfDivN, 2022-05-18:
### Variant

USART

### Control type

Commutation

### Description

hoverboard 
firmware: hack-foc_hover + uart3

I assembled the firmware to receive control data via uart3, but I noticed that when turning with the gyroscope connected, the platform rolls out further than the required angle.
This is due to the fact that there is some lag between the Send commands(int16_t steer, int16_t speed). Perhaps there is a way to reduce the delay between consecutive commands. I ask for help in this matter. What should I do?

aLfDivN, 2022-05-19:
[https://cloud.mail.ru/public/KgYK/USGzcVkL1](url) that's my robot

aLfDivN, 2022-05-26:
Now for stop motor i'll send many commands
  for (ii=0;ii<20;ii++)  {
    Send(0, 1);
    delay(6);                 }
but it's not that accuracy how i want :(

Candas1, 2022-05-26:
Hi,

If you read the arduino code, you will see there is a time_send interval that you can adjust(100ms).

The foc main loop runs every 5ms, this can be reduced in config.h if needed. 

Also check the wiki page about the parameters, there is a FILTER parameter that can introduce a lag for your use case.

aLfDivN, 2022-06-13:
I found in the wiki that the Speed Limit parameter will help me
The DEFAULT_RATE parameter is used to limit the rate of change of input data in the main loop between 2 iterations of the loop.

Candas1, 2022-06-19:
👍

closed 2022-06-19 by Candas1