EFeru/hoverboard-firmware-hack-FOC #305: SWD Flashing while battery is disconnected

RoboEye, 2022-05-19:
### Description

Does anyone know why 3.3V from st-link during firmware flashing is a bad idea, schematics and latching power circuit doesn't show anything about this "forbidden scenario" and the circuit only appears to work when the battery is connected since the enable signal is applied to TIP127 to power the whole circuit up and the 3.3v from "SWD" header seems to be connected directly to the 3.3v line on the PCB !?

RoboEye, 2022-08-04:
No One to Answer!!

Candas1, 2022-08-04:
Hi,

I don't answer because I don't know the answer. This was an advice coming from the original repository.
Altough it might work most of the time, it seems some people bricked there mainboards.

There is not much people active here, you can try your chance in the Telegram group as well if you really need an answer.

yurka6600, 2022-08-04:
Hi.
I do flash using power from st-link adapter, but with disconnected battery and "empty" capacitors, no issues yet.

EFeru, 2022-08-04:
With battery disconnected and capacitors discharged i see no issue. I did that multiple times.

juodumas, 2022-08-05:
I've bricked one board like this. After you flash a firmware, you also want to test it and for that you need to connect the battery. And then if the test fails you want to flash again...