EFeru/hoverboard-firmware-hack-FOC #311: Electric bike twist throttle

Germany111, 2022-05-24:
### Description

Who is the best variant to run this throttle it has reverse and 3 speed buttons

Candas1, 2022-05-26:
Hi,

For throttle and brake, you should use the hovercar variant, please read the wiki page that describes how it works. 

The firmware doesn't handle speed/reverse buttons, you would have to adapt the code to your needs.

JSphere22, 2022-05-29:
[image]

I can not get this thumb throttle to work. I can not get a single pot to work. I have it wired for speed only and have commented out steering. I have been reading the wiki for months now. Can some help point me in a direction that might help or where to purchase a thumb throttle that works.

closed 2023-03-01 by Candas1