EFeru/hoverboard-firmware-hack-FOC #312: 4WD simultaneous wheels turn

MarosMotaj, 2022-05-24:
### Variant

USART

### Description

Hi All,

I have 4WD platform with two hoverboards (RC receiver connected via PWM to Arduino UNO. From Arduino UART connection with board1 and board2). Me question is, it is possible to setting up simultaneous wheel turn? Because now when I move the RC transmitter, one hoverboard is faster. Thanks :)

Candas1, 2022-05-26:
Hi,

Why don't you use speed control mode?

MarosMotaj, 2022-05-26:
Hi,

Where can I find it in wiki?

Thanks :)

MarosMotaj, 2022-05-28:
Can you send me please, how to do that? Thanks :)

Candas1, 2022-05-29:
> Can you send me please, how to do that? Thanks :)

Please check config.h file

MarosMotaj, 2022-05-29:
ok, thanks :)

MarosMotaj, 2022-05-29:
I found it, but how to activate SPD_MODE? I have to comment other mods??

#define COM_CTRL        0               // [-] Commutation Control Type
#define SIN_CTRL        1               // [-] Sinusoidal Control Type
#define FOC_CTRL        2               // [-] Field Oriented Control (FOC) Type

#define OPEN_MODE       0               // [-] OPEN mode
#define VLT_MODE        1               // [-] VOLTAGE mode
#define SPD_MODE        2               // [-] SPEED mode
#define TRQ_MODE        3               // [-] TORQUE mode

Candas1, 2022-05-29:
> I found it, but how to activate SPD_MODE? I have to comment other mods??
>
> #define COM_CTRL        0               // [-] Commutation Control Type
> #define SIN_CTRL        1               // [-] [...]

You need to change it here
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4b16b647716ddc7ae03b0674b3b83bf304642c8f/Inc/config.h#L151

MarosMotaj, 2022-05-29:
Thanks, I set: #define CTRL_MOD_REQ    SPD_MODE
Then I flashed the FW to both boards but the speeds was still diferent...
I configure on ESP32 harware UART ports, first for one board and second to other board and set diferent speed constants regarding on wheel speeds and it is quite good.

Candas1, 2022-06-02:
👍

closed 2022-06-02 by Candas1