EFeru/hoverboard-firmware-hack-FOC #313: unable to turn off, until max speed

dexop, 2022-06-02:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

hello, thanks for all your work on this project.
i have problems turning off the main board.
dual mode, left side ADC, right IBUS.

everithing is working just fine, adc side, calibration, ibus side also.
the problem occurs when i try to shut down main board on button. it doesn't turn off.
but, when i push throtle (adc or ibus) to the max, then button react, and main board shuts down.

any ideas what to check?

Candas1, 2022-06-03:
Hi,

What battery are you using?

dexop, 2022-06-03:
standard battery  from hover board. 36v  4400mah

Candas1, 2022-06-03:
How is your board looking like ?
Have you checking in the [wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility) if it's a standard one ?

Candas1, 2022-06-03:
What I don't understand is the button works when you are at max speed ? or the board turns off at max speed ?

dexop, 2022-06-04:
button works only at max speed given by ibus or adc.

Candas1, 2022-06-04:
Well then I don't know what this could come from.
If you are able to enter auto-calibration, this proves the button works and is mapped the right MCU pin, so turning off should work as well.

The reason I asked those question is because:
* some boards have the button mapped to a different pin, you would still be able to turn the board on, but you wouldn't be able to enter autocalibration or poweroff
* if the battery voltage is too low, because of the voltage divider between the button and the mcu, the voltage could be too low to pick up, but it would also impact autocalibration and poweroff 
* if the tip127 mosfet is shorted, the board would be always ON and you wouldn't be able to poweroff at all, but it doesn't seem to be your case

mrendu43, 2022-06-13:
Hello.. 
Have you menaget to solve this problem, i  have the same with mine board using usart3.
I can power on, and enter calibration but it wont power off.

Candas1, 2022-07-04:
> Hello.. Have you menaget to solve this problem, i have the same with mine board using usart3. I can power on, and enter calibration but it wont power off.

Have you checked what I suggested ?

lsls5921, 2022-07-16:
Is there any way to use the hover board to control the DC brush motor

Candas1, 2022-07-20:
> Is there any way to use the hover board to control the DC brush motor

Your questions is not relevant in this issue.
and the answer is no...

hainanlin, 2022-09-12:
Is there any way to use the hover board to control the DC brush motor

Candas1, 2022-09-13:
The anwer is still no.... not with this firmware
But if you want you can code it

closed 2022-12-23 by Candas1