EFeru/hoverboard-firmware-hack-FOC #316: Free wheel on Sinusoidal

Misiaa28, 2022-06-04:
### Control type

Sinusoidale

### Control mode

Speed

### What can we do to make the firmware better?

Introduce the freewheel function for all modes. After releasing the handle, the wheel can turn freely. In Sinusoidal mode, high speeds can be obtained with a lower amperage, but the freewheel does not work. It would satisfy the users of scooters, bicycles and skateboards.

### Describe suggestions or alternatives you have considered

Introduce the freewheel function for all modes. After releasing the handle, the wheel can turn freely. In Sinusoidal mode, high speeds can be obtained with a lower amperage, but the freewheel does not work. It would satisfy the users of scooters, bicycles and skateboards.

hbnewbie, 2023-01-07:
i would also love to see this if its possible. it would be ideal for people making scooters or gokarts for themselves and trying to get higher speeds.

abyy28y, 2023-02-01:
it is good idea

Candas1, 2023-02-02:
How would you guys brake then ?

Candas1, 2023-02-02:
Then field weakening should be improved with foc...

abyy28y, 2023-02-02:
I’m talking about the speed and smooth running of wheel

hbnewbie, 2023-02-04:
i did a little searching and it looks like eferu has said freewheeling not possible in modes other than trq mode, due to different operating principles https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/212

Candas1, 2023-02-04:
Not even available on vesc
https://vesc-project.com/node/134