EFeru/hoverboard-firmware-hack-FOC #322: Is it possible to disable the brake of one wheel out of two ?

1UserOK1, 2022-06-18:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

Hello. I have a homemade all-wheel drive scooter. There was a problem setting up the brake.
Auto-calibration is successful, but after applying the brake, the front wheel spins forward, not backward.
I checked the hall sensors and tried to swap them (green, blue, yellow). It didn't work.
I would like to disable the braking of one wheel in the firmware without disabling the wheel completely from work, but I don’t know if this is possible?
Please excuse my bad English, it's not my native language.
Best regards, Andrey

Candas1, 2022-06-18:
Hi,

That's probably one of the issues I am describing [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#toolbox-troubleshooting)
I understand what you are trying to do but this is not fixing the root cause
I think the phase and halls are matching, but the sequence is not right.
What happens if you swap 2 phases and halls of the same color?
You might have to use a parameter to revert the direction of the wheel after that.

1UserOK1, 2022-06-18:
I tried to find a working combination by swapping the phase wires and the outputs of the hall sensors, but unfortunately I could not achieve normal operation.
Either the wheel does not rotate, or it rotates with jerks. None of the combinations work.
The second wheel is connected in the same way, it works wonderfully and stably, but the front one does not.
I checked the signal from the hall sensors on the front wheel with an oscilloscope, it is there.
So far, I'm still leaning towards a malfunction of one or more hall sensors.
I don't know where to look for the reason.

Candas1, 2022-06-18:
Have you tried swapping the wheels?

1UserOK1, 2022-06-18:
No, I haven't tried. Now swapping wheels is not very quick and easy.
I connected the rear wheel through a different connector on the hall sensor (homemade), they are no longer interchangeable.
But may have to check the wheel, as you advise, in order to find the culprit

1UserOK1, 2022-06-18:
It is a pity that you can not turn off the brake of one wheel. This would solve my problem most quickly and painlessly,
 When driving, both wheels behave normally, only the "strange front wheel" seems to vibrate and make a little more noise, but not much

1UserOK1, 2022-06-19:
I was able to resolve the issue. Instead of ADC mode, I chose Hovercare mode, calibrated and everything became normal. As a brake handle, I have a regular button. It is necessary to shunt the output with a capacitor and a 1-10K resistor
I don't know why, but it worked.
maybe this will help someone

closed 2022-06-19 by 1UserOK1

Candas1, 2022-06-19:
Ok I didn't see you were using the adc variant. Throttle and brakecare only supported with the hovercar variant. In adc variant this input is steering