EFeru/hoverboard-firmware-hack-FOC #324: brake mode

landrisan, 2022-06-27:
### Variant

USART

### Control type

FOC

### Control mode

Torque

### Description

I would like to use the motor as an electromagnetic brake, with adjustable gain. So, would be possible instead of setting speed over UART, to set up the gain of the brake and the motor to not rotate at all... just wait to be rotated and give a brake feedback according to the value recived via UART

Candas1, 2022-07-04:
Hi,

It's not possible with the firmware but you could do this yourself from an arduino.
In torque mode you are sending a torque setpoint via UART, and you know at what speed the wheel is rotating.

landrisan, 2022-07-05:
ok, I can do that. In fact, I already tryed. But for high braking gains, the motor must run in reverse. For example, if the motor is rotated to the left, to brake-it, I must send command to rotate to the right. Is ok to work like that? For long term ussage, the mainboard can work like this? I also noticed that the input voltage is increased, from 36v to 40v, hat was with battery tested. If I need to use with a power supply, I think I must use a Zenner diode and a power resistor to drain over voltage. Will test..