EFeru/hoverboard-firmware-hack-FOC #326: E-bike with 2 motor goes in reverse

Sekai1942, 2022-07-02:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

Hi. i tried to make the an ebike with simple thumb throttle but it goes only in reverse.
I tried hovercar variant but it does not spin the motors.
As for the brake part i would like to use a different thumb throttle.
And i do not need steering at all but if i put it to 0 the motor dont spin in adc variant or hovercar.
I auto calibrated the thumb throttle and it works great but only in reverse.
I have the 100k on both signals.
3.3V its ok.
Is the a way to find why it goes in reverse?
And i tried to put an i2c lcd 16x2 on right cable but it does not work at all.
Thanks

> // Define to prevent recursive inclusion
> #ifndef CONFIG_H
> #define CONFIG_H
> 
> #include "stm32f1xx_hal.h"
> 
> // ############################### VARIANT SELECTION ###############################
> // PlatformIO: uncomment desired variant in platformio.ini
> // Keil uVision: select desired variant from the Target drop down menu (to the right of the Load button)
> // Ubuntu: define the desired build variant here if you want to use make in console
> // or use VARIANT environment variable for example like "make -e VARIANT=VARIANT_NUNCHUK". Select only one at a time.
> #if !defined(PLATFORMIO)
>   #define VARIANT_ADC         // Variant for control via ADC input
>   //#define VARIANT_USART       // Variant for Serial control via USART3 input
>   //#define VARIANT_NUNCHUK     // Variant for Nunchuk controlled vehicle build
>   //#define VARIANT_PPM         // Variant for RC-Remote with PPM-Sum Signal
>   //#define VARIANT_PWM         // Variant for RC-Remote with PWM Signal
>   //#define VARIANT_IBUS        // Variant for RC-Remotes with FLYSKY IBUS
>   //#define VARIANT_HOVERCAR    // Variant for HOVERCAR build
>   //#define VARIANT_HOVERBOARD  // Variant for HOVERBOARD build
>   //#define VARIANT_TRANSPOTTER // Variant for TRANSPOTTER build https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter https://hackaday.io/project/161891-transpotter-ng
>   //#define VARIANT_SKATEBOARD  // Variant for SKATEBOARD build
> #endif
> // ########################### END OF VARIANT SELECTION ############################
> 
> 
> // ############################### DO-NOT-TOUCH SETTINGS ###############################
> #define PWM_FREQ            16000     // PWM frequency in Hz / is also used for buzzer
> #define DEAD_TIME              48     // PWM deadtime
> #ifdef VARIANT_TRANSPOTTER
>   #define DELAY_IN_MAIN_LOOP    2
> #else
>   #define DELAY_IN_MAIN_LOOP    5     // in ms. default 5. it is independent of all the timing critical stuff. do not touch if you do not know what you are doing.
> #endif
> #define TIMEOUT                20     // number of wrong / missing input commands before emergency off
> #define A2BIT_CONV             50     // A to bit for current conversion on ADC. Example: 1 A = 50, 2 A = 100, etc
> // #define PRINTF_FLOAT_SUPPORT          // [-] Uncomment this for printf to support float on Serial Debug. It will increase code size! Better to avoid it!
> 
> // ADC conversion time definitions
> #define ADC_CONV_TIME_1C5       (14)  //Total ADC clock cycles / conversion = (  1.5+12.5)
> #define ADC_CONV_TIME_7C5       (20)  //Total ADC clock cycles / conversion = (  7.5+12.5)
> #define ADC_CONV_TIME_13C5      (26)  //Total ADC clock cycles / conversion = ( 13.5+12.5)
> #define ADC_CONV_TIME_28C5      (41)  //Total ADC clock cycles / conversion = ( 28.5+12.5)
> #define ADC_CONV_TIME_41C5      (54)  //Total ADC clock cycles / conversion = ( 41.5+12.5)
> #define ADC_CONV_TIME_55C5      (68)  //Total ADC clock cycles / conversion = ( 55.5+12.5)
[... 719 lines cut ...]
> 
> #if defined(SUPPORT_BUTTONS_RIGHT) && (defined(CONTROL_NUNCHUK) || defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT) || defined(DEBUG_I2C_LCD))
>   #error SUPPORT_BUTTONS_RIGHT and (CONTROL_NUNCHUK or CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT or DEBUG_I2C_LCD) not allowed. It is on the same cable.
> #endif
> 
> #if defined(CONTROL_NUNCHUK) && (defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT) || defined(DEBUG_I2C_LCD))
>   #error CONTROL_NUNCHUK and (CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT or DEBUG_I2C_LCD) not allowed. It is on the same cable.
> #endif
> 
> #if defined(DEBUG_I2C_LCD) && (defined(CONTROL_PPM_RIGHT) || defined(CONTROL_PWM_RIGHT))
>   #error DEBUG_I2C_LCD and (CONTROL_PPM_RIGHT or CONTROL_PWM_RIGHT) not allowed. It is on the same cable.
> #endif
> 
> #if defined(CONTROL_PPM_RIGHT) && defined(CONTROL_PWM_RIGHT)
>   #error CONTROL_PPM_RIGHT and CONTROL_PWM_RIGHT not allowed. It is on the same cable.
> #endif
> 
> 
> // Functional checks
> #if (defined(CONTROL_PPM_LEFT) || defined(CONTROL_PPM_RIGHT)) && !defined(PPM_NUM_CHANNELS)
>   #error Total number of PPM channels needs to be set
> #endif
> // ############################# END OF VALIDATE SETTINGS ############################
> 
> #endif

Candas1, 2022-07-02:
Hi,

Have you read the wiki? Specially [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR#rocket-features)
Also check the parameters if you want to revert one wheel's rotation

Sekai1942, 2022-07-02:
i read it. i did break to 0 input but the motor dont spin.

Candas1, 2022-07-03:
but it's not 0 in what you have shared
[image]

You need to make brake input type 0 if you want to be able to use the auto calibration with hovercar variant without a brake throttle. You need to connect the throttle to the right wire also.

Sekai1942, 2022-07-04:
i put it back so that i can check if the brake works.
the big problem is when i press the throttle it goes in reverse.

Candas1, 2022-07-04:
I already gave you the solution for that, you need to check the parameters in the wiki or in hovercar variant in config.h

Sekai1942, 2022-07-04:
i tried that but the motor dont spin.
But now i seem to freeze the motherboard.
it cannot detect the board at all.
I tried reset method but it does not work.Still no detection.
I have a second motherboard that works so the stlink v2 works.

Sekai1942, 2022-07-07:
ok. i dont know what happen but the motherboard is back.
Like i said only reverse is working if i disable the steering then the motor dont spin at all.
Even if i auto calibrate nothing happens.
Any ideas?

Candas1, 2022-07-07:
Is it so hard to read a single page completely?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Parameters#rewind-invert-wheel-direction

Sekai1942, 2022-07-07:
thanks for the link it help.
I have both a pot and hall sensor so either of them can be a throttle as a test.
i put STEER_COEFFICIENT to 0 and DEFAULT_SPEED_COEFFICIENT to 16384 and now both motor spin with no reverse function activated.
The next problem is the speed is not increasing. i have the max rpm to 1000.It just stays at 300rpm (measured with a laser speedometer)
#define N_MOT_MAX       1000

Candas1, 2022-07-07:
Well you need to read again
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits

I won't help more

closed 2022-07-07 by Candas1

Sekai1942, 2022-07-10:
thanks for the info it works great.I had to experiment with load because with no load nothing works right.
Anyway i have a issue with the amperage being to high. i get 40A at start and I reduce the Imax to 10A.
Is the I max for single motor or dual?
At full speed at 40 km/h it gets to 56A.

Candas1, 2022-07-10:
It's literally mentioned in the first sentence 😜

Sekai1942, 2022-07-10:
yes i understand how to it works but what i am asking if the dc curent is not monitored by the board how can i limit it

> To avoid both current limitation to compete when using FOC, set I_DC_MAX to I_MOT_MAX + 2A.
Some boards are missing the components to measure the DC current, this means current limiting won't work with Commutation and Sinusoidal, FOC control type should be used in this case.

I am using foc control but the amper limitation is not working.

Candas1, 2022-07-10:
Have you use auto calibration?

Sekai1942, 2022-07-10:
yes i used it

Candas1, 2022-07-10:
Well then again you didn't fully tead the last link I shared