EFeru/hoverboard-firmware-hack-FOC #328: Using 12V lead acid battery

js333031, 2022-07-07:
### Description

I don't have the 36V Li Ion batteries but do have the mainboard and motors/wheels. Is it possible to:
1. Flash new firmware w/o power source connected to the XT60 connector? The firmware flashing instructions seem to allow so but older documentation I believe required ">36V" power connected to XT60
2. Run the firmware and motors w/12V lead acid battery?

I happen to have the Pb battery and wanted to check before I purchase Li battery.

Candas1, 2022-07-20:
Hi,

You can power the chip with 3.3v from stlink, but it's not recommended, it seems few people ended up bricking the board.
I am not sure running on 12v is a good idea, the power button will for sure not work with that low voltage.

RoboDurden, 2022-07-29:
The mosfet gate drivers run on 14V. but technically they should already work at 12V.

derFrickler, 2023-01-21:
Is there a way to enable the board without the power button? just turning on when plugging power?
Want to try to run the board on a 4S lipo

RoboDurden, 2023-01-21:
Shorting the first TIP127 TO220 transistor will always power the board. This happens when that transistor blows. These boards then will always turn on once the battery is connected.
Have a look at the schematic.pdf ..

You should uncomment 
```
  if (inactivity_timeout_counter > (INACTIVITY_TIMEOUT * 60 * 1000) / (DELAY_IN_MAIN_LOOP + 1)) {  // rest of main loop needs maybe 1ms
      poweroff();
    }
```
in main.c because the power down melody might hang at the last tone :-(

derFrickler, 2023-01-21:
Great, thanks a lot, I will try that

derFrickler, 2023-01-21:
Seems its not an TIP127 on my board (the 2nd on is), first one is an IRF9530N.
But bridging source and drain did the Job ;_)

Thanks a lot!

RoboDurden, 2023-01-21:
Yes, some boards have a mosfet instead of a to220 transistor. Not really neccessary. Don't forget to post a youtube video when you have something nice to show :-)

abyy28y, 2023-02-01:
does it work for u derfrickler?

derFrickler, 2023-02-01:
YES, works great, i will do a video soon.

abyy28y, 2023-02-01:
works on 12 volt?

abyy28y, 2023-02-01:
> Shorting the first TIP127 TO220 transistor will always power the board. This happens when that transistor blows. These boards then will always turn on once the battery is connected. Have a look at [...]

excuse me is there is any way or tip127 substitute to make the board powered on 12 volt using the power button?

RoboDurden, 2023-02-01:
> excuse me is there is any way or tip127 substitute to make the board powered on 12 volt using the power button?

The power button triggers the littler transistor Q1 which will activate that first TIP127.
You might need to decrease the resistance of R1 from 30 kOhm to maybe 20k or 10k in order to trigger Q1 when V_bat is only 12V instead of 36V.
See the schematics pages 5 and 6 of https://raw.githubusercontent.com/EFeru/hoverboard-firmware-hack-FOC/main/docs/20150722_hoverboard_sch.pdf

You could try to feed in the 12V with a 10k Ohm resistor directly into the On_Switch Jumper J1 pin 1..
Just a quick idea..

I am currently working on the Gen2 boards: https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x

abyy28y, 2023-02-01:
Thank u so much RoboDurden and good luck, I will try it.

ash25mi, 2023-03-19:
Hi, any news on using a 12V battery ?

derFrickler, 2023-03-21:
Its working on 4s Lipo, so it should be fine at 12V aswell, just need to set the batery low levels in the code.

derFrickler, 2023-03-21:
BTW, i promised a  video, heres one of the first test-drives: 
https://www.youtube.com/watch?v=epiRJ1xkm4k

derFrickler, 2023-03-21:
I bought the tracks years ago from Aliexpress for $40, but they are not available anymore (at least not from Europe).
Power is fine like this as its only a fun vehicle and i am only using 4S right now. It was able to pull the Sled with my 2 small kids on the Snow...

RoboDurden, 2023-03-21:
I don't like fun projects... 
Better make it a $200 12s suicide gps bomber tank to support Ukrainian soldiers :-/

ash25mi, 2023-03-21:
> Its working on 4s Lipo, so it should be fine at 12V aswell, just need to set the batery low levels in the code.
Thanks for the feedback ! Nice build.

> I forgot to tell you that I auto-power-on my solar camper with a 10 uF Capacitor and a ? KOhm discharge resistor. When powering the board, the capacitor is discharged and simulates a short button push :-) That way you eliminate the big spark when plugging in the battery. Even 10A Power switches do not last long because of the big capacitors of the hoverboard controller that I only use to generate up to 2 amps pedal assist.
Thanks it is a good addition !

> I don't like fun projects... Better make it a $200 12s suicide gps bomber tank to support Ukrainian soldiers :-/
Small drones in Ukraine are also consumer grade, sometimes based on open-source projects, and also using hobby components.
If you want to see it that way, the existence of hobbies enables people to learn skills and create projects, and an industry to exist.

ash25mi, 2023-03-21:
> Its working on 4s Lipo, so it should be fine at 12V aswell, just need to set the batery low levels in the code.
Thanks for the feedback ! Nice build.

> I forgot to tell you that I auto-power-on my solar camper with a 10 uF Capacitor and a ? KOhm discharge resistor. When powering the board, the capacitor is discharged and simulates a short button [...]

Thanks it is a good addition !

> I don't like fun projects... Better make it a $200 12s suicide gps bomber tank to support Ukrainian soldiers :-/

Small drones in Ukraine are also consumer grade, sometimes based on open-source projects, and also using hobby components.
If you want to see it that way, the existence of hobbies enables people to learn skills and create projects, and an industry to exist.

TinkerWorX, 2023-12-11:
Just wanted to share a quick example for anyone looking how to quickly make the board turn off with lower voltage. For my setup I wanted it to run off 18v Ryobi batteries which wasn't letting it power. I added a 22k ohm resistor as shown in the image. This drops the R1 resistance from 30k to 12.6k. (ignore the soldering job, hard to apply enough heat on the massive copper area)
[image]
It now powers up from 18v without issues. Remember to update the firmware as well to avoid it shutting down from under-voltage detection.

RoboDurden, 2023-12-11:
Be aware that now attaching a fully loaded 36V battery might exceed the 3.3V/5V of the MCU input pin !

TinkerWorX, 2023-12-11:
Good point, I added a warning to my post to make sure people understand this.

thecesco, 2024-06-29:
> Shorting the first TIP127 TO220 transistor will always power the board. This happens when that transistor blows. These boards then will always turn on once the battery is connected. Have a look at [...]

Hi! Could you please help how to short the tip? Thanks!

RoboDurden, 2024-06-29:
If you don't know how to google and read the TIP127 datasheet, you should not shortcut any electronics.

derFrickler, 2025-09-15:
As stated above, i shorted the TIP127/IRF9530N next to the switch and commented the poweroff line.
That worked fine for me.

BTW, thats where the tracked platform ended up ;_)
https://der-frickler.net/spass/hoverbob

On the page you find some of my other hoverboard projects, like the 4wd quad i did lately.