EFeru/hoverboard-firmware-hack-FOC #331: is it possible to drive the motors with a pwm and a dir line

SynthetikzZ, 2022-08-06:
### Control type

FOC

### Control mode

Torque

### Description

is it possible to provide the controller with a pwm signal for motor strength and a direction signal (0,1)  from another stm board to control the motor?

Candas1, 2022-08-06:
Hi,

No, the firmware doesn't support this.

SynthetikzZ, 2022-08-06:
oh i read trough the source code right now, i think i can use the Variant steering with some adjustments

Reponja, 2022-08-23:
Did you get it to work? I need that functionality for my Force Feedback Simracing wheel i am making out of hoverboard

closed 2023-01-31 by Candas1

korvolle, 2023-07-07:
> Did you get it to work? I need that functionality for my Force Feedback Simracing wheel i am making out of hoverboard

Did you managed to make it work?

Reponja, 2023-07-08:
> > Did you get it to work? I need that functionality for my Force Feedback Simracing wheel i am making out of hoverboard
> 
> Did you managed to make it work?

Nah, i got myself an Odrive