EFeru/hoverboard-firmware-hack-FOC #336: Bobycar mode: Throttle/Break mapped to wrong input

xilefn, 2022-08-24:
### Variant

ADC

### Control type

Commutation

### Control mode

Voltage

### What is the bug and how to reproduce it ?

When pressing the pedals not the ADC Values of the pedals changes but the Voltage and Temperature reeding.

I flashed a new board with the firmaware configured for Bobycar. 
After connecting throttle/break pedal and calibrating them they it not work (wheels spin slowly all the time)
After looking at the Serial output of the beard i noticed that when pressing the throttle pedal the batterie voltage changes and when pressing the break pedal the temperature changes. 

I measured the board and the wires are connected to the right uC pins (16,17). Thats why the only thing I can think of is a wrong mapping somewhere in the software. 

I am still trying to find where to change the ADC mapping in the code but i didn´t find it yet. 

My board is the 2017.06.01 PHC version. with a GD32F103 controller. 

Both pedals not pressed:
in1:1693 in2:0 cmdL:115 cmdR:115 BatADC:1044 BatV:3823 TempADC:1335 Temp:983
Both pedals pressed:
in1:1680 in2:0 cmdL:114 cmdR:114 BatADC:3190 BatV:7781 TempADC:3034 Temp:-2338

Candas1, 2022-08-29:
Hi,

When you say configured for bobbycar, you mean it's a different fork ?

Candas1, 2022-08-29:
You should better discuss it [here](https://github.com/larsmm/hoverboard-firmware-hack-FOC-bbcar/issues/3)

closed 2022-09-12 by Candas1

LONG-9527, 2023-03-08:
没办法用

LONG-9527, 2023-03-10:
@Candas1