EFeru/hoverboard-firmware-hack-FOC #339: Compatibility with Jetson main board

humble800, 2022-08-28:
### Variant

USART

### Control type

FOC

### Control mode

Speed

### Description

I wonder if board members have been able to port the FOC source code to Jeston V5 Active Balance E-Scooter, and/or know the issues with constant, rapid, high pitch beeping after the main board is turn on.

I have been able to compile and run the FOC firmware on an older main board like the one pictured on [Wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility).  Thank you so much for developing this code.

However, I wasn't able run the same code on the main board donated from Jeston V5 Active Balance E-Scooter using the alternative board variant (BOARD_VARIANT = 1 in the config.h).  The board has a STM32F103RCT6 controller.  The motors do spin under FOC speed mode and under sinusoidal modes.  I have two problems: 
1. The main board emits rapid, constant, high pitch beeps as soon as the main board is turned on.  The beep patterns are different from the [diagnostic beep patterns](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics).  
2. The other issue is that the UART debug output appears to indicate that the main board is not getting the correct rpm read out.  The rpm readouts alternate between the correct speed and 0.

I tested the connection using multi meter, it appears that the Jeston main board has connected the left and right hall to the same cpu pins as the older version of the main board pictured on [Wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility). 

Does anyone happen to know if this board is compatible with the FOC software?  Any advice will be appreciated.  

Below are the picture of the donor hover board and the pictures of the main board.

[image]
[image]
[image]

Candas1, 2022-08-29:
Hi,

Can you share a recording of the beeps you are getting ?

humble800, 2022-08-30:
Hi there- Many thanks for your willingness to help taking a look.  Here is a short video clips showing the rapid buzzer beeping noise after the board is on.  I am controlling the rpm using FOC speed mode and am able to change the rpm of both wheels.  I am just not able to figure out what part of the code is causing the rapid beeping.

https://user-images.githubusercontent.com/88641966/187320090-89c4c4fc-b545-499e-844a-10cf33b7ffb7.mp4

I actually also saw another user (Chaos99) here that had modified the code to make it work with the G3 version of the main board https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/90.  I compared their changes with the FOC code, it appears that setting BOARD_VARIANT = 1 in the config.h would have incorporated all of Chaos99's changes.

Any advice will be appreciated.

Candas1, 2022-08-30:
Yes, I contributed with this alternate board variant.
Those rapid beeps are not normal, it's the first time I hear it.
It almost sounds like backwards beeps but faster. 
Can you try disabling backward beeps in config.h? 
Understanding what condition triggers the beeps will help.

humble800, 2022-08-30:
Thank you very again for the suggestion.

I did make the change you suggest and I believe the beep is from the code that I added to check whether the board is connected to the charger and if it is, it beeps and powers off for safety reasons so that it doesn't run away with chargers plugged in.  After I commented out the code, there's no more buzz sound.

The code snip is below in util.c.  This works with old main board, meaning when the charger is plugged in, the board will poweroff().  However this same piece code doesn't work with the BOARD_VARIANT = 1.  I know the CHARGER_PIN differs between the two board variants.  I wonder if the HAL_GPIO_ReadPin(CHARGER_PORT, CHARGER_PIN) result is different between boards.  When the old main board is plugged in, HAL_GPIO_ReadPin(CHARGER_PORT, CHARGER_PIN) returns false; but the same function returns true for the G3 main board when it is plugged in the charger.

//====defines.h
#if BOARD_VARIANT == 0
#define CHARGER_PIN GPIO_PIN_12
#define CHARGER_PORT GPIOA
#elif BOARD_VARIANT == 1
#define CHARGER_PIN GPIO_PIN_11
#define CHARGER_PORT GPIOA
#endif

//====util.c
void poweroffPressCheck(void) {
  if (!(HAL_GPIO_ReadPin(CHARGER_PORT, CHARGER_PIN)))
  {
    //beepShort(5);
    //poweroff();
  }
//the rest of the codes are omitted

Candas1, 2022-08-30:
I am trying to help you figure out where the beeps come from, but you modified the code to add beeps...
I should have started from there.

Candas1, 2022-08-30:
The alternate pinout was provided by someone else, I don't have such board.
And the charger pin is not used for any feature in this firmware.
So it could be the pin for the charger is wrong.

humble800, 2022-08-31:
Thank you for taking a look at this.  I appreciate it.  We can close this question now, though I haven't figured out why yet.

Separately, I am curious if you happen to know what the extra white wire by the hall sensors cable is used for.  Is it for the hub motor temperature sensing or something else?  Thanks again.

Candas1, 2022-08-31:
Yes I think it's to measure motor temperature, but it's not handled in the firmware, only few have this variant of the board.

closed 2022-08-31 by humble800