EFeru/hoverboard-firmware-hack-FOC #346: Accurate control of left and right wheel speed.

ChristopheLaurent, 2022-10-12:
### Variant

USART

### Control type

FOC

### Control mode

Speed

### Description

First of all, I wish to thank Emanuel Feru for this incredible work. I consider an application where I would control the speed of each wheel accurately from another microcontroller. I have a couple of related questions:

1/ Reading the code, I guess that one wheel will run at speed + steer and the other wheel at speed - steer, where speed and steer are specified as integer RPMs in USART frame. So basically, if I set the speed to 10 and the steer to 10 in my USART command, one of the wheel should be blocked while the other wheel should run at 20 RPM, am I right ?

2/ I also noticed a smoothing filter on speed and steer, that is tunable with a set of parameters in the config.h file line 184:

`/* FILTER is in fixdt(0,16,16): VAL_fixedPoint = VAL_floatingPoint * 2^16. In this case 6553 = 0.1 * 2^16
 * Value of COEFFICIENT is in fixdt(1,16,14)
 * If VAL_floatingPoint >= 0, VAL_fixedPoint = VAL_floatingPoint * 2^14
 * If VAL_floatingPoint < 0,  VAL_fixedPoint = 2^16 + floor(VAL_floatingPoint * 2^14).
*/
// Value of RATE is in fixdt(1,16,4): VAL_fixedPoint = VAL_floatingPoint * 2^4. In this case 480 = 30 * 2^4
#define DEFAULT_RATE                480   // 30.0f [-] lower value == slower rate [0, 32767] = [0.0, 2047.9375]. Do NOT make rate negative (>32767)
#define DEFAULT_FILTER              6553  // Default for FILTER 0.1f [-] lower value == softer filter [0, 65535] = [0.0 - 1.0].
#define DEFAULT_SPEED_COEFFICIENT   16384 // Default for SPEED_COEFFICIENT 1.0f [-] higher value == stronger. [0, 65535] = [-2.0 - 2.0]. In this case 16384 = 1.0 * 2^14
#define DEFAULT_STEER_COEFFICIENT   8192  // Defualt for STEER_COEFFICIENT 0.5f [-] higher value == stronger. [0, 65535] = [-2.0 - 2.0]. In this case  8192 = 0.5 * 2^14. If you do not want any steering, set it to 0.
`
Which values would disable this filter, assuming that the other microcontroller already filtered speed and steer upfront by another mean ?

3/ Eventually, USART outputs the speed of each wheel. However, it seems it is not the real speed computed from hall sensors, but the required speed after the filter. Is it possible to get the real speed evaluated from hall sensors ? To which extent does it differ from the required speed if I use Speed Control Mode ?

Best Regards

Candas1, 2022-10-12:
Hi,

Please check the wiki

ChristopheLaurent, 2022-10-12:
Hi, I did spend a couple of hours on the wiki already, but I probably missed the answers to my specific questions. Could you please tell me which page page explain the filter coefficient, or the difference between real speed vs control speed depending on Control Mode ? Best Regards

ChristopheLaurent, 2022-10-15:
Hi, sure, I did check this page more than once. But the Input Command Structure description lacks a couple of details. It does not say that Speed unit is RPM, assuming that speed factor in configuration is 1.0. It is even less obvious that Steer unit is also RPM, and represent a difference of speed between left and right wheel. I had to read the code to understand it, but maybe I'm wrong so I did ask my 1st question with a numeric example. There is not much in the wiki about the low pass filter parameters as well, so I think my questions were legitimate. Anyway, I will get a second hand board very soon, I will experiment it by myself, and I may propose documentation improvements after my tests if you don't mind.

Candas1, 2022-10-15:
Don't get me wrong, I am willing to help, but I think it's all documented there, probably I am bad at documenting then.

Why are you making your life complicated with speed and steer if you want to control each wheel individually:
[image]

Is this not describing filtering good enough ?
[image]

The feedback you get is real speed in RPM:
[image]

In Speed mode, the input is the target speed, I can probably add it's also RPMs
[image]

Candas1, 2022-10-25:
Oui c'est bien ça 😜

closed 2022-10-27 by ChristopheLaurent

ChristopheLaurent, 2022-10-31:
Hi Candas,
I've made some progress on my project and I can now drive the hoverboard motors from an ESP32. I did not manage to use SoftwareSerial for reading but HardwareSerial works fine. However, I'm puzzled by the [feedback data](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART#speech_balloon-feedback) . Here is what I get asking for a constant speed of 100 RPM using the same print as in [hoverserial.ino](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/main/Arduino/hoverserial/hoverserial.ino) :
`...
1: 0 2: 100 3: -27 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -27 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -27 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 29 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 29 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 29 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -28 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -28 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -26 4: 28 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -27 4: 25 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -27 4: 25 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -24 4: 29 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -24 4: 29 5: 3980 6: 243 7: 0
1: 0 2: 100 3: -24 4: 28 5: 3980 6: 243 7: 0
...`
Cmd2 is 100 RPM as expected, the battery voltage is a little bit higher than expected (I measured about 37V instead of 39.8V) and the temperature of 24.3 degC looks to be fine. However, what is the unit of the speed in the range [24, 29] ?  I understand that the sign is due to the rotation direction, but is there a mix of "revolutions per minute" and "radians per minute" somewhere ?

Candas1, 2022-10-31:
As I told you before. SpeedR and SpeedL are in rpm

Candas1, 2022-10-31:
In the form you said you use speed control mode, is it really the case...

ChristopheLaurent, 2022-10-31:
Thanks a lot for your help ! I missed that CTRL_MOD_REQ parameter in config.h and I was still in VLT_MODE. The control loop works as expected with SPD_MODE.

AlfretzNehemia, 2023-07-17:
Hi guys thank you for this discussion. But right now, I want to know If we select VOLT_MODE for motor control mode. The measured left and right speed doesn't get same value as the input?

Amitkor12, 2024-02-07:
Hi guys, I have a quick question about the steering part of the wheels .
If I need to steer one wheel with more speed than the other, let's say a 2:1 ratio, that is, the left will spin twice the right.
For this what changes should be made?