EFeru/hoverboard-firmware-hack-FOC #348: PCB maximum battery voltage

Japla74, 2022-10-18:
### Variant

ADC

### Control type

FOC

### Control mode

Torque

### Description

Hi, I happen to have two NMC battery packs, both max 29,2V. I wonder, that will PCB survive if I connect them in series to supply max 58,4V to the howerboard PCB? I checked that main caps are 63V and fets up to 100V, but is there some unknown limiting factors?

I am creating quad version with rubber tracks, aiming for quite high peak output power.
[image]

Candas1, 2022-10-24:
Hi,

I think it's too close, you would need to upgrade to 100v caps probably.

emo8899, 2023-02-28:
> If caps is only issue, those I can replace easily. What about internal voltages, smaller ones are done with 7805 etc, but is there any else which would raise too high? Thanks!

hi  Japla
  Have you solved this problem? Can we use a higher voltage?

Candas1, 2023-02-28:
Check the wiki

closed 2023-03-01 by Candas1